osrf/ros2_test_cases #1207: Intra-process executables

marcoag, 2024-04-29:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Links

* [Intra-process demos](https://github.com/ros2/demos/blob/jazzy/intra_process_demo/README.md)

## Process

    
            
### The executables in the README are present and work as expected.

samuel-ongzx, 2024-05-14:
## Process:
- [ ]The executables in the README are present and work as expected.

## Method:
- Clone & build [demos repo](https://github.com/ros2/demos/tree/jazzy) on `jazzy` branch
- Verify according to [link](https://github.com/ros2/demos/blob/jazzy/intra_process_demo/README.md)

## Result:
- [x] Two Node Pipeline (`ok`)
[image]
- [x] Cyclic Pipeline (`ok`)
[image]

Hooking up a webcam later tonight to test below nodes, but if anybody wants to chime in, feel free!
- [ ] Image Pipeline All in One (`TODO`)
- [ ] Image Pipeline All Separately (`TODO`)
- [ ] Image Pipeline With Two Image Views (`TODO`)

samuel-ongzx, 2024-05-15:
Waiting for PR to be merged: https://github.com/ros2/demos/pull/689

samuel-ongzx, 2024-05-15:
Updated PR from ros2/demos: https://github.com/ros2/demos/pull/690

clalancette, 2024-05-15:
Because we have the open PR, I'm going to go ahead and close this one.  Thank you for testing!

closed 2024-05-15 by clalancette