osrf/ros2_test_cases #1210: Quality of Service Demos

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Links

* [Demo URL](https://github.com/ros2/demos/blob/jazzy/quality_of_service_demo/README.md)

## Process

    
            
### Step 1 - Deadline with pause
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 2000
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 2000
```
                
    

    
        
#### Expected Result
        
        
The publisher will publish for 5 seconds, then pause for 2 - deadline events will start firing on both participants, until the publisher comes back.
                
    
            
### Step 2 - Deadline no pause
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 0
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 0
```
                
    

    
        
#### Expected Result
        
        
The publisher doesn't actually pause, no deadline misses should occur.
                
    
            
### Step 3 - Lifespan, some survive
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp lifespan 1000 --publish-count 10 --subscribe-after 3000
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py lifespan 1000 --publish-count 10 --subscribe-after 3000
```
                
    

    
        
#### Expected Result
        
        
After a few seconds, you should see (approximately) messages 4-9 printed from the subscriber.

The first few messages, with 1 second lifespan, were gone by the time the subscriber joined after 3 seconds.

                
    
            
### Step 4 - Lifespan, all survive
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp lifespan 4000 --publish-count 10 --subscribe-after 3000
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py lifespan 4000 --publish-count 10 --subscribe-after 3000
```
                
    

    
        
#### Expected Result
        
        
After a few seconds, you should see all of the messages (0-9) printed from the subscriber.

All messages, with their 4 second lifespan, survived until the subscriber joined.

                
    
            
### Step 5 - Liveliness
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp liveliness 1000 --kill-publisher-after 2000
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py liveliness 1000 --kill-publisher-after 2000
```
                
    

    
        
#### Expected Result
        
        
After 2 seconds, the publisher will be killed, and the subscriber will receive a callback 1 second after that notifying it that the liveliness has changed.
                
    
            
### Step 6 - Incompatible QoS Offered/Requested
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp incompatible_qos durability
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py incompatible_qos durability
```
                
    

    
        
#### Expected Result
        
        
The publisher in this demo will publish 5 messages, but also output a notification stating that its offered QoS is incompatible with the subscriber that is also created by this demo. The subscriber in this demo will similarly output a notification stating that its requested QoS is incompatible with the QoS offered by the publisher.

                
    
            
### Step 7 - Interactive Quality of Service Demo
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp interactive_publisher
```
                
```bash
# StdIn - terminal 2
ros2 run quality_of_service_demo_cpp interactive_subscriber
```
                
In the publisher's terminal, press `p` to manually assert the liveliness of the publisher, for when Liveliness kind is `MANUAL_BY_TOPIC`.
                
In the publisher's terminal, press `s` to toggle enabling/disabling of publishing messages
                
In the publisher and subscriber's terminal, press `q` to print the current QoS policy.
                
In the publisher and subscriptions's terminal, press `x` to stop and exit the demo.
                
    

    
        
#### Expected Result
        
        
All key-press commands work.

MGupta28, 2024-05-03:
Step 1 - Deadline with pause

- For cpp
 [image]
- For python
[image]

Step 2 - Deadline no pause
- For cpp
[image]

- For python
[image]

Step 3 - Lifespan, some survive

- For cpp
[image]

- For python
[image]

Step 4 - Lifespan, all survive

- For cpp
[image]

- For python
[image]

Step 5 - Liveliness

- For cpp
[image]

- For python
[image]

Step 6 - Incompatible QoS Offered/Requested

- For cpp
[image]

- For python
[image]

Step 7 - Interactive Quality of Service Demo

- Publisher side outputs
1. Pressing `p`
[image]

2. Pressing `s`
[image]

3. Pressing `q`
[image]

4. Pressing `x`
Demo was stopped

- For Subscriber
1. Pressing `q`
[image]

2. Pressing `x`
Demo was stopped

### Conclusion

All the steps were working as expected.

clalancette, 2024-05-06:
Thank you for testing!

closed 2024-05-06 by clalancette