osrf/ros2_test_cases #1211: Quality of Service Demos

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Links

* [Demo URL](https://github.com/ros2/demos/blob/jazzy/quality_of_service_demo/README.md)

## Process

    
            
### Step 1 - Deadline with pause
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 2000
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 2000
```
                
    

    
        
#### Expected Result
        
        
The publisher will publish for 5 seconds, then pause for 2 - deadline events will start firing on both participants, until the publisher comes back.
                
    
            
### Step 2 - Deadline no pause
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 0
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 0
```
                
    

    
        
#### Expected Result
        
        
The publisher doesn't actually pause, no deadline misses should occur.
                
    
            
### Step 3 - Lifespan, some survive
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp lifespan 1000 --publish-count 10 --subscribe-after 3000
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py lifespan 1000 --publish-count 10 --subscribe-after 3000
```
                
    

    
        
#### Expected Result
        
        
After a few seconds, you should see (approximately) messages 4-9 printed from the subscriber.

The first few messages, with 1 second lifespan, were gone by the time the subscriber joined after 3 seconds.

                
    
            
### Step 4 - Lifespan, all survive
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp lifespan 4000 --publish-count 10 --subscribe-after 3000
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py lifespan 4000 --publish-count 10 --subscribe-after 3000
```
                
    

    
        
#### Expected Result
        
        
After a few seconds, you should see all of the messages (0-9) printed from the subscriber.

All messages, with their 4 second lifespan, survived until the subscriber joined.

                
    
            
### Step 5 - Liveliness
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp liveliness 1000 --kill-publisher-after 2000
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py liveliness 1000 --kill-publisher-after 2000
```
                
    

    
        
#### Expected Result
        
        
After 2 seconds, the publisher will be killed, and the subscriber will receive a callback 1 second after that notifying it that the liveliness has changed.
                
    
            
### Step 6 - Incompatible QoS Offered/Requested
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp incompatible_qos durability
```
                
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_py incompatible_qos durability
```
                
    

    
        
#### Expected Result
        
        
The publisher in this demo will publish 5 messages, but also output a notification stating that its offered QoS is incompatible with the subscriber that is also created by this demo. The subscriber in this demo will similarly output a notification stating that its requested QoS is incompatible with the QoS offered by the publisher.

                
    
            
### Step 7 - Interactive Quality of Service Demo
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp interactive_publisher
```
                
```bash
# StdIn - terminal 2
ros2 run quality_of_service_demo_cpp interactive_subscriber
```
                
In the publisher's terminal, press `p` to manually assert the liveliness of the publisher, for when Liveliness kind is `MANUAL_BY_TOPIC`.
                
In the publisher's terminal, press `s` to toggle enabling/disabling of publishing messages
                
In the publisher and subscriber's terminal, press `q` to print the current QoS policy.
                
In the publisher and subscriptions's terminal, press `x` to stop and exit the demo.
                
    

    
        
#### Expected Result
        
        
All key-press commands work.

Barry-Xu-2018, 2024-05-09:
I would like to handle this test.

Barry-Xu-2018, 2024-05-09:
Test environment:
- OS: Ubuntu Noble 24.04 (Docker)
- Chip set: AMD64 (Intel i7-13700K)
- Build type: Debian
- DDS: CycloneDDS

### Step 1 - Deadline with pause
- cpp
[image]

- py
[image]

The results met expectations, the test passed. :white_check_mark: 

### Step 2 - Deadline no pause
- cpp
[image]

- py
[image]

The results met expectations, the test passed. :white_check_mark: 

### Step 3 - Lifespan, some survive
- cpp
[image]

- py
[image]

The results met expectations, the test passed. :white_check_mark: 

### Step 4 - Lifespan, all survive

- cpp
[image]

- py
[image]

The results met expectations, the test passed. :white_check_mark: 

### Step 5 - Liveliness
- cpp
[image]

- py
[image]

The results met expectations, the test passed. :white_check_mark: 

### Step 6 - Incompatible QoS Offered/Requested
- cpp
[image]

- py
[image]

The results met expectations, the test passed. :white_check_mark: 

### Step 7 - Interactive Quality of Service Demo

After launching publisher, and press `p`
[image]

Press `s` and then press `s` again
[image]

Press `q` on publisher terminal
[image]

Press `q` on subscriber terminal
[image]

Press `x` on publisher terminal
[image]

Press `x` on subscriber terminal
[image]

The results met expectations, the test passed. :white_check_mark: 

### Conclusion

All test steps passed.  :white_check_mark:

Barry-Xu-2018, 2024-05-09:
@ahcorde Test finished. The test results meet expectations.

marcoag, 2024-05-09:
Great, thanks for testing!

closed 2024-05-09 by marcoag