osrf/ros2_test_cases #1212: Lifecycle demo

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Links

* [Reference URL](https://github.com/ros2/demos/blob/jazzy/lifecycle/README.rst)

## Process

    
            
### Run files individually
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run lifecycle lifecycle_talker
```
                
[image]
                
```bash
# StdIn - terminal 2
ros2 run lifecycle lifecycle_listener
```
                
[image]
                
```bash
# StdIn - terminal 3
ros2 run lifecycle lifecycle_service_client
```
                
[image]
                
    

    
        
#### Expected Result
        
        
The images match.

Nikhil-Singhal-06, 2024-05-02:
Could you please assign this issue to me? 

Additionally, The expected result images aren't showing up for me. Is it the same for you as well?

clalancette, 2024-05-02:
> Additionally, The expected result images aren't showing up for me. Is it the same for you as well?

Yeah, you are right.  Those images aren't working for some reason.  Just go ahead and run the examples and paste the output here.

Nikhil-Singhal-06, 2024-05-06:
This works as expected.

[image]

closed 2024-05-06 by ahcorde