osrf/ros2_test_cases #1214: Topic monitor

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Links

* [Reference URL](https://github.com/ros2/demos/blob/jazzy/topic_monitor/README.md)

## Process

    
            
### Comparing reliability QoS settings
            
        

    
    

    
        
            
#### Actions
            
        
        
You will need two machines with ROS 2: one mobile and one stationary.

                
```bash
# StdIn - terminal 1
ros2 run topic_monitor topic_monitor --display
```
                
Run the `ros2 launch topic_monitor reliability_demo_launch.py` executable on the stationary machine. This will start two nodes: one publishing in “reliable” mode, and one in “best effort”.

                
```bash
# StdIn - terminal 1
ros2 launch topic_monitor reliability_demo_launch.py
```
                
Start the monitor on a mobile machine such as a laptop. Use `ros2 run topic_monitor topic_monitor --display --allowed-latency 5` to account for any latency that may occur re-sending the reliable messages.

                
```bash
# StdIn - terminal 1
ros2 run topic_monitor topic_monitor --display --allowed-latency 5
```
                
Take the mobile machine out of range of the monitor, and observe how the reception rates differ for the different topics.
                
    

    
        
#### Expected Results
        
        
You should see that the "reliable" topic has a reception rate that is almost always either 0 or 100%, while the "best effort" topic has a reception rate that fluctuates based on the strength of the connection.

                
[image]

fujitatomoya, 2024-05-09:
@marcoag since i do not have 2 machines, i tried to test this with 2 containers.

it seems that works without any problems, but plotting is different from expected result. i think we need a wifi network which is kinda unstable connectivity? this is as much as i can test, hope this helps if nobody else takes this with 2 physical machines.

[image]

marcoag, 2024-05-09:
@fujitatomoya thanks for testing! as  you mentioned, your output makes sense, will leave this open though in case anyone else can test with two machines.

samuel-ongzx, 2024-05-15:
Just tested this with 2 machines!

## Details
Setup:
- DDS: FastDDS
- Build type: Debian (apt-get)
- OS: Docker: `ghcr.io/sloretz/ros-testing:jazzy-desktop`
- Chip set: AMD64 (1 desktop & 1 laptop)

## Verification

Had to walk out of my house + 50m to get the drop in signal with some stops to try and get the best effort 😮‍💨 .
[image]

ahcorde, 2024-05-15:
Thank you for testing and for moving around

closed 2024-05-15 by ahcorde