osrf/ros2_test_cases #1218: Topic executables in `demo_nodes_py`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `listener` and `talker`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_py talker
```
                
```bash
# StdIn - terminal 2
ros2 run demo_nodes_py listener
```
                
    

    
    
            
### Step 2 - Check `listener` and `talker` QOS
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_py talker_qos
```
                
```bash
# StdIn - terminal 2
ros2 run demo_nodes_py listener_qos
```
                
    

    
    
            
### Step 3 - Check serialized `listener` and `talker`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_py talker
```
                
```bash
# StdIn - terminal 2
ros2 run demo_nodes_py listener_serialized
```

fujitatomoya, 2024-05-03:
@marcoag 

test result,

```bash
<details>
  <summary>Step 1 - Check listener and talker</summary>

```bash
root@51cdd59e1f3e:~# ros2 run demo_nodes_py talker
[INFO] [1714712057.238006017] [talker]: Publishing: "Hello World: 0"
[INFO] [1714712058.227442811] [talker]: Publishing: "Hello World: 1"
[INFO] [1714712059.227407865] [talker]: Publishing: "Hello World: 2"
[INFO] [1714712060.227390356] [talker]: Publishing: "Hello World: 3"
[INFO] [1714712061.227416047] [talker]: Publishing: "Hello World: 4"

root@51cdd59e1f3e:~# ros2 run demo_nodes_py listener
[INFO] [1714712057.246999429] [listener]: I heard: [Hello World: 0]
[INFO] [1714712058.229579223] [listener]: I heard: [Hello World: 1]
[INFO] [1714712059.229319556] [listener]: I heard: [Hello World: 2]
[INFO] [1714712060.229399902] [listener]: I heard: [Hello World: 3]
[INFO] [1714712061.229390558] [listener]: I heard: [Hello World: 4]
```

</details>

<details>
  <summary>Step 2 - Check listener and talker QOS</summary>

```bash
root@51cdd59e1f3e:~ros2 run demo_nodes_py talker_qosos
[INFO] [1714712076.436599899] [talker_qos]: Best effort talker
[INFO] [1714712077.437842805] [talker_qos]: Publishing: "Hello World: 0"
[INFO] [1714712078.439083255] [talker_qos]: Publishing: "Hello World: 1"
[INFO] [1714712079.439088583] [talker_qos]: Publishing: "Hello World: 2"
[INFO] [1714712080.439069136] [talker_qos]: Publishing: "Hello World: 3"
[INFO] [1714712081.439084983] [talker_qos]: Publishing: "Hello World: 4"
[INFO] [1714712082.439067385] [talker_qos]: Publishing: "Hello World: 5"

root@51cdd59e1f3e:~ros2 run demo_nodes_py listener_qosos
[INFO] [1714712075.382722527] [listener_qos]: Best effort listener
[INFO] [1714712077.438465031] [listener_qos]: I heard: [Hello World: 0]
[INFO] [1714712078.441163977] [listener_qos]: I heard: [Hello World: 1]
[INFO] [1714712079.441082053] [listener_qos]: I heard: [Hello World: 2]
[INFO] [1714712080.441064366] [listener_qos]: I heard: [Hello World: 3]
[INFO] [1714712081.441027830] [listener_qos]: I heard: [Hello World: 4]
[INFO] [1714712082.441049590] [listener_qos]: I heard: [Hello World: 5]
```

</details>

<details>
  <summary>Step 3 - Check serialized listener and talker</summary>

```bash
root@51cdd59e1f3e:~ros2 run demo_nodes_py talkerer
[INFO] [1714712100.001472281] [talker]: Publishing: "Hello World: 0"
[INFO] [1714712100.991663533] [talker]: Publishing: "Hello World: 1"
[INFO] [1714712101.991627193] [talker]: Publishing: "Hello World: 2"
[INFO] [1714712102.991638898] [talker]: Publishing: "Hello World: 3"
[INFO] [1714712103.991659519] [talker]: Publishing: "Hello World: 4"

root@51cdd59e1f3e:~ros2 run demo_nodes_py listener_serializeded
[INFO] [1714712100.010925848] [serialized_subscriber]: I heard: "b'\x00\x01\x00\x00\x0f\x00\x00\x00Hello World: 0\x00\x00'"
[INFO] [1714712100.993778914] [serialized_subscriber]: I heard: "b'\x00\x01\x00\x00\x0f\x00\x00\x00Hello World: 1\x00\x00'"
[INFO] [1714712101.993598283] [serialized_subscriber]: I heard: "b'\x00\x01\x00\x00\x0f\x00\x00\x00Hello World: 2\x00\x00'"
[INFO] [1714712102.993658148] [serialized_subscriber]: I heard: "b'\x00\x01\x00\x00\x0f\x00\x00\x00Hello World: 3\x00\x00'"
[INFO] [1714712103.993568999] [serialized_subscriber]: I heard: "b'\x00\x01\x00\x00\x0f\x00\x00\x00Hello World: 4\x00\x00'"
```

</details>
```

no problem observed, we can close this.

marcoag, 2024-05-03:
Thanks a lot!

closed 2024-05-03 by marcoag