osrf/ros2_test_cases #1222: Parameter executables in `demo_nodes_py`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Async parameter client
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp parameter_blackboard
```
                
```bash
# StdIn - terminal 2
ros2 run demo_nodes_py async_param_client
```

fujitatomoya, 2024-05-03:
@marcoag 

test result,

```bash
root@51cdd59e1f3e:~# ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [1714711157.911230824] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '6' parameters already!

root@51cdd59e1f3e:~# ros2 run demo_nodes_py async_param_client
[INFO] [1714711163.557886306] [async_param_client]: Setting parameters:
[INFO] [1714711163.558624870] [async_param_client]:     int_parameter:
[INFO] [1714711163.558825810] [async_param_client]:         successful: True
[INFO] [1714711163.559035753] [async_param_client]:     bool_parameter:
[INFO] [1714711163.559249775] [async_param_client]:         successful: True
[INFO] [1714711163.559443295] [async_param_client]:     string_parameter:
[INFO] [1714711163.559639389] [async_param_client]:         successful: True
[INFO] [1714711163.559888477] [async_param_client]: Listing Parameters:
[INFO] [1714711163.560391493] [async_param_client]:     - bool_parameter
[INFO] [1714711163.560623473] [async_param_client]:     - int_parameter
[INFO] [1714711163.560841214] [async_param_client]:     - string_parameter
[INFO] [1714711163.561076081] [async_param_client]: Getting parameters:
[INFO] [1714711163.561601988] [async_param_client]:     - int_parameter: 10
[INFO] [1714711163.561846208] [async_param_client]:     - bool_parameter: False
[INFO] [1714711163.562105696] [async_param_client]:     - string_parameter: Fee Fi Fo Fum
[INFO] [1714711163.562351382] [async_param_client]: Loading parameters:
[INFO] [1714711163.564519913] [async_param_client]:     other_int_parameter:
[INFO] [1714711163.564804864] [async_param_client]:         successful: True
[INFO] [1714711163.565104297] [async_param_client]:         value: 0
[INFO] [1714711163.565367404] [async_param_client]:     int_parameter:
[INFO] [1714711163.565651318] [async_param_client]:         successful: True
[INFO] [1714711163.565928754] [async_param_client]:         value: 12
[INFO] [1714711163.566192524] [async_param_client]:     string_parameter:
[INFO] [1714711163.566474214] [async_param_client]:         successful: True
[INFO] [1714711163.566759718] [async_param_client]:         value: I smell the blood of an Englishman
[INFO] [1714711163.567032255] [async_param_client]:     other_string_parameter:
[INFO] [1714711163.567307904] [async_param_client]:         successful: True
[INFO] [1714711163.567589611] [async_param_client]:         value: One fish, two fish, Red fish, blue fish
[INFO] [1714711163.567863759] [async_param_client]:     bool_parameter:
[INFO] [1714711163.568141173] [async_param_client]:         successful: True
[INFO] [1714711163.568421227] [async_param_client]:         value: False
[INFO] [1714711163.568710125] [async_param_client]: Deleting parameters:
[INFO] [1714711163.569352079] [async_param_client]:     other_int_parameter:
[INFO] [1714711163.569670430] [async_param_client]:         successful: True
[INFO] [1714711163.569982121] [async_param_client]:         reason:
[INFO] [1714711163.570308617] [async_param_client]:     other_string_parameter:
[INFO] [1714711163.570626398] [async_param_client]:         successful: True
[INFO] [1714711163.570944792] [async_param_client]:         reason:
[INFO] [1714711163.571250030] [async_param_client]:     string_parameter:
[INFO] [1714711163.571560189] [async_param_client]:         successful: True
[INFO] [1714711163.571883332] [async_param_client]:         reason:
```

no problem observed, we can close this.

marcoag, 2024-05-03:
Great, thanks for testing!

closed 2024-05-03 by marcoag