osrf/ros2_test_cases #1226: Executables in `quality_of_service_demo_cpp`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `lifespan`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp lifespan
```
                
    

    
    
            
### Step 2 - Check `liveliness`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp liveliness
```
                
    

    
    
            
### Step 3 - Check `deadline`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run quality_of_service_demo_cpp deadline
```

fujitatomoya, 2024-05-03:
@marcoag 

test result,

```bash
root@51cdd59e1f3e:~# ros2 run quality_of_service_demo_cpp lifespan 1
[INFO] [1714710956.952594795] [talker]: Talker starting up
[INFO] [1714710956.953052558] [listener]: Listener starting up
[INFO] [1714710957.453305638] [talker]: Publishing: 'Talker says 0'
[INFO] [1714710957.953322793] [talker]: Publishing: 'Talker says 1'
[INFO] [1714710958.453339427] [talker]: Publishing: 'Talker says 2'
[INFO] [1714710958.953301455] [talker]: Publishing: 'Talker says 3'
[INFO] [1714710959.453331983] [talker]: Publishing: 'Talker says 4'
^C[INFO] [1714710981.555379989] [rclcpp]: signal_handler(signum=2)

root@51cdd59e1f3e:~# ros2 run quality_of_service_demo_cpp liveliness 2
[INFO] [1714710985.663813975] [talker]: Talker starting up
[INFO] [1714710985.664191199] [listener]: Listener starting up
[INFO] [1714710986.164445548] [talker]: Publishing: 'Talker says 0'
[INFO] [1714710986.164829382] [listener]: Listener heard: [Talker says 0]
[INFO] [1714710986.164899140] [listener]: Liveliness changed event:
[INFO] [1714710986.164918224] [listener]:   alive_count: 1
[INFO] [1714710986.164936128] [listener]:   not_alive_count: 0
[INFO] [1714710986.164952669] [listener]:   alive_count_change: 1
[INFO] [1714710986.164969728] [listener]:   not_alive_count_change: 0
[INFO] [1714710986.166159640] [listener]: Liveliness changed event:
[INFO] [1714710986.166277276] [listener]:   alive_count: 0
[INFO] [1714710986.166297646] [listener]:   not_alive_count: 1
...<snip>

root@51cdd59e1f3e:~# ros2 run quality_of_service_demo_cpp deadline 3
[INFO] [1714711035.575886899] [talker]: Talker starting up
[INFO] [1714711035.576296473] [listener]: Listener starting up
[INFO] [1714711036.076463789] [talker]: Publishing: 'Talker says 0'
[INFO] [1714711036.076836421] [listener]: Listener heard: [Talker says 0]
[INFO] [1714711036.079057026] [talker]: Offered deadline missed - total 1 delta 1
[INFO] [1714711036.079171494] [listener]: Requested deadline missed - total 1 delta 1
[INFO] [1714711036.081146103] [talker]: Offered deadline missed - total 2 delta 1
[INFO] [1714711036.081227652] [listener]: Requested deadline missed - total 2 delta 1
[INFO] [1714711036.083320889] [talker]: Offered deadline missed - total 3 delta 1
[INFO] [1714711036.083334667] [listener]: Requested deadline missed - total 3 delta 1
...<snip>
```

no problem observed, we can close this.

marcoag, 2024-05-03:
Thanks a lot!

closed 2024-05-03 by marcoag