osrf/ros2_test_cases #1232: Executables in `tf2_ros`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `tf2_echo`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run tf2_ros tf2_echo
```
                
    

    
    
            
### Step 2 - Check `tf2_monitor`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run tf2_ros tf2_monitor
```
                
    

    
    
            
### Step 3 - Check `static_transform_publisher`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run tf2_ros static_transform_publisher --frame-id world --child-frame-id map
```

Nikhil-Singhal-06, 2024-05-02:
I tested this one and the executables work as expected.

[image]

clalancette, 2024-05-02:
Thank you for testing!

closed 2024-05-02 by clalancette