osrf/ros2_test_cases #1233: Executables in `tf2_ros`

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Process

    
            
### Step 1 - Check `tf2_echo`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run tf2_ros tf2_echo
```
                
    

    
    
            
### Step 2 - Check `tf2_monitor`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run tf2_ros tf2_monitor
```
                
    

    
    
            
### Step 3 - Check `static_transform_publisher`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run tf2_ros static_transform_publisher --frame-id world --child-frame-id map
```

mikaelarguedas, 2024-05-05:
All tests passed: :heavy_check_mark:

## Step 1 - Check tf2_echo

:heavy_check_mark: test passed

```
$ ros2 run tf2_ros tf2_echo world turtle1
[INFO] [1714915210.411318644] [tf2_echo]: Waiting for transform world ->  turtle1: Invalid frame ID "world" passed to canTransform argument target_frame - frame does not exist
At time 1714915211.398563490
- Translation: [5.544, 5.544, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
- Rotation: in RPY (radian) [0.000, -0.000, 0.000]
- Rotation: in RPY (degree) [0.000, -0.000, 0.000]
- Matrix:
  1.000  0.000  0.000  5.544
  0.000  1.000  0.000  5.544
  0.000  0.000  1.000  0.000
  0.000  0.000  0.000  1.000
At time 1714915212.389828212
- Translation: [5.544, 5.544, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
- Rotation: in RPY (radian) [0.000, -0.000, 0.000]
- Rotation: in RPY (degree) [0.000, -0.000, 0.000]
- Matrix:
  1.000  0.000  0.000  5.544
  0.000  1.000  0.000  5.544
  0.000  0.000  1.000  0.000
  0.000  0.000  0.000  1.000
```

## Step 2 - Check tf2_monitor

:heavy_check_mark: test passed

```
$ ros2 run tf2_ros tf2_monitor world turtle1
[INFO] [1714915218.029073540] [tf2_monitor_main]: Waiting for transform world ->  turtle1: Invalid frame ID "world" passed to canTransform argument target_frame - frame does not exist
Gathering data on world -> turtle1 for 10 seconds...

RESULTS: for world to turtle1
Chain is: world -> turtle1
Net delay     avg = 0.00165528: max = 0.0163763

Frames:
Frame: turtle1, published by <no authority available>, Average Delay: 0.000565342, Max Delay: 0.000917673

All Broadcasters:
Node: <no authority available> 125.256 Hz, Average Delay: 0.000556188 Max Delay: 0.000712395

RESULTS: for world to turtle1
Chain is: world -> turtle1
Net delay     avg = 0.00283961: max = 0.0163763

Frames:
Frame: turtle1, published by <no authority available>, Average Delay: 0.000556552, Max Delay: 0.000717878

All Broadcasters:
Node: <no authority available> 125.248 Hz, Average Delay: 0.000558466 Max Delay: 0.000718832
```

## Step 3 - Check static_transform_publisher

```
$ ros2 run tf2_ros static_transform_publisher --frame-id world --child-frame-id map
[INFO] [1714915263.625956866] [static_transform_publisher_2yLQoRrIr0cPM9S6]: Spinning until stopped - publishing transform
translation: ('0.000000', '0.000000', '0.000000')
rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
from 'world' to 'map'
```

```
At time 0.0
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
- Rotation: in RPY (radian) [0.000, -0.000, 0.000]
- Rotation: in RPY (degree) [0.000, -0.000, 0.000]
- Matrix:
  1.000  0.000  0.000  0.000
  0.000  1.000  0.000  0.000
  0.000  0.000  1.000  0.000
  0.000  0.000  0.000  1.000
At time 0.0
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
- Rotation: in RPY (radian) [0.000, -0.000, 0.000]
- Rotation: in RPY (degree) [0.000, -0.000, 0.000]
- Matrix:
  1.000  0.000  0.000  0.000
  0.000  1.000  0.000  0.000
  0.000  0.000  1.000  0.000
  0.000  0.000  0.000  1.000
```

marcoag, 2024-05-06:
Thanks for testing!

closed 2024-05-06 by marcoag