osrf/ros2_test_cases #1242: Executables in `examples_rclcpp_minimal_composition`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `composition_composed`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_composition composition_composed
```
                
    

    
    
            
### Step 2 - Check `composition_publisher`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_composition composition_publisher
```
                
    

    
    
            
### Step 3 - Check `composition_subscriber`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_composition composition_subscriber
```

fujitatomoya, 2024-05-03:
@marcoag 

test result.

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclcpp_minimal_composition composition_composed
[INFO] [1714710160.742223338] [publisher_node]: Publisher: 'Hello, world! 0'
[INFO] [1714710160.742678658] [subscriber_node]: Subscriber: 'Hello, world! 0'
[INFO] [1714710161.242181722] [publisher_node]: Publisher: 'Hello, world! 1'
[INFO] [1714710161.242448636] [subscriber_node]: Subscriber: 'Hello, world! 1'
[INFO] [1714710161.742241129] [publisher_node]: Publisher: 'Hello, world! 2'
[INFO] [1714710161.742514355] [subscriber_node]: Subscriber: 'Hello, world! 2'
[INFO] [1714710162.242212687] [publisher_node]: Publisher: 'Hello, world! 3'
[INFO] [1714710162.242486845] [subscriber_node]: Subscriber: 'Hello, world! 3'
[INFO] [1714710162.742227614] [publisher_node]: Publisher: 'Hello, world! 4'
[INFO] [1714710162.742506094] [subscriber_node]: Subscriber: 'Hello, world! 4'
[INFO] [1714710163.242185459] [publisher_node]: Publisher: 'Hello, world! 5'
[INFO] [1714710163.242464062] [subscriber_node]: Subscriber: 'Hello, world! 5'
^C[INFO] [1714710163.695147314] [rclcpp]: signal_handler(signum=2)
root@51cdd59e1f3e:~# ros2 run examples_rclcpp_minimal_composition composition_publisher
[INFO] [1714710169.463469081] [publisher_node]: Publisher: 'Hello, world! 0'
[INFO] [1714710169.963466644] [publisher_node]: Publisher: 'Hello, world! 1'
[INFO] [1714710170.463458024] [publisher_node]: Publisher: 'Hello, world! 2'
[INFO] [1714710170.963461108] [publisher_node]: Publisher: 'Hello, world! 3'
[INFO] [1714710171.463491765] [publisher_node]: Publisher: 'Hello, world! 4'
[INFO] [1714710171.963464466] [publisher_node]: Publisher: 'Hello, world! 5'
^C[INFO] [1714710172.227499652] [rclcpp]: signal_handler(signum=2)
root@51cdd59e1f3e:~# ros2 run examples_rclcpp_minimal_composition composition_subscriber
[INFO] [1714710183.859926215] [subscriber_node]: Subscriber: 'Hello, world! 0'
[INFO] [1714710184.359703524] [subscriber_node]: Subscriber: 'Hello, world! 1'
[INFO] [1714710184.859693215] [subscriber_node]: Subscriber: 'Hello, world! 2'
[INFO] [1714710185.359690755] [subscriber_node]: Subscriber: 'Hello, world! 3'
[INFO] [1714710185.859744301] [subscriber_node]: Subscriber: 'Hello, world! 4'
[INFO] [1714710186.359734302] [subscriber_node]: Subscriber: 'Hello, world! 5'
^C[INFO] [1714710187.722058725] [rclcpp]: signal_handler(signum=2)
```

no problem observed, we can close this.

marcoag, 2024-05-03:
Thanks for the test!

closed 2024-05-03 by marcoag