osrf/ros2_test_cases #1244: Executables in `examples_rclcpp_minimal_action_server`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `action_server_not_composable`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_server action_server_not_composable
```
                
    

    
    
            
### Step 2 - Check `action_server_member_functions`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
```

fujitatomoya, 2024-05-03:
@marcoag i can take care of this.

fujitatomoya, 2024-05-03:
test result.

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclcpp_minimal_action_server action_server_not_composable
[INFO] [1714710020.204590152] [server]: Got goal request with order 10
[INFO] [1714710020.204714814] [server]: Executing goal
[INFO] [1714710020.204823430] [server]: Publish Feedback
[INFO] [1714710021.204978727] [server]: Publish Feedback
[INFO] [1714710022.205065086] [server]: Publish Feedback
[INFO] [1714710023.205063825] [server]: Publish Feedback
[INFO] [1714710024.204944269] [server]: Publish Feedback
[INFO] [1714710025.204967038] [server]: Publish Feedback
[INFO] [1714710026.205047060] [server]: Publish Feedback
[INFO] [1714710027.205010145] [server]: Publish Feedback
[INFO] [1714710028.205029554] [server]: Publish Feedback
[INFO] [1714710029.205130686] [server]: Goal Succeeded
^C[INFO] [1714710031.825691583] [rclcpp]: signal_handler(signum=2)
root@51cdd59e1f3e:~# ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
[INFO] [1714710039.256476228] [minimal_action_server]: Received goal request with order 10
[INFO] [1714710039.256655846] [minimal_action_server]: Executing goal
[INFO] [1714710039.256709968] [minimal_action_server]: Publish Feedback
[INFO] [1714710040.256925890] [minimal_action_server]: Publish Feedback
[INFO] [1714710041.257036859] [minimal_action_server]: Publish Feedback
[INFO] [1714710042.256905701] [minimal_action_server]: Publish Feedback
[INFO] [1714710043.257000478] [minimal_action_server]: Publish Feedback
[INFO] [1714710044.256984360] [minimal_action_server]: Publish Feedback
[INFO] [1714710045.257098098] [minimal_action_server]: Publish Feedback
[INFO] [1714710046.256982205] [minimal_action_server]: Publish Feedback
[INFO] [1714710047.257012296] [minimal_action_server]: Publish Feedback
[INFO] [1714710048.257103538] [minimal_action_server]: Goal Succeeded
^C[INFO] [1714710050.186576061] [rclcpp]: signal_handler(signum=2)
```

no problem observed, we can close this.

marcoag, 2024-05-03:
Thanks for the test!

closed 2024-05-03 by marcoag