osrf/ros2_test_cases #1246: Executables in `examples_rclcpp_minimal_action_client`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `action_client_not_composable`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_client action_client_not_composable
```
                
    

    
    
            
### Step 2 - Check `action_client_not_composable_with_cancel`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_cancel
```
                
    

    
    
            
### Step 3 - Check `action_client_not_composable_with_feedback`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_feedback
```
                
    

    
    
            
### Step 4 - Check `action_client_member_functions`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_client action_client_member_functions
```

fujitatomoya, 2024-05-03:
@marcoag i can take care of this.

fujitatomoya, 2024-05-03:
test result.

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclcpp_minimal_action_client action_client_not_composable
[INFO] [1714709876.248514413] [minimal_action_client]: Sending goal
[INFO] [1714709876.248771223] [minimal_action_client]: Waiting for result
[INFO] [1714709885.249601115] [minimal_action_client]: result received
[INFO] [1714709885.249693741] [minimal_action_client]: 0
[INFO] [1714709885.249713352] [minimal_action_client]: 1
[INFO] [1714709885.249728093] [minimal_action_client]: 1
[INFO] [1714709885.249743074] [minimal_action_client]: 2
[INFO] [1714709885.249758161] [minimal_action_client]: 3
[INFO] [1714709885.249774339] [minimal_action_client]: 5
[INFO] [1714709885.249789462] [minimal_action_client]: 8
[INFO] [1714709885.249811648] [minimal_action_client]: 13
[INFO] [1714709885.249827285] [minimal_action_client]: 21
[INFO] [1714709885.249840981] [minimal_action_client]: 34
[INFO] [1714709885.249861890] [minimal_action_client]: 55
root@51cdd59e1f3e:~# ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_cancel
[INFO] [1714709898.086509182] [minimal_action_client]: Sending goal
[INFO] [1714709901.087077050] [minimal_action_client]: canceling goal
[INFO] [1714709901.088322641] [minimal_action_client]: goal is being canceled
[INFO] [1714709901.088401664] [minimal_action_client]: Waiting for result
[ERROR] [1714709902.087971941] [minimal_action_client]: Goal was canceled
[ros2run]: Process exited with failure 1
root@51cdd59e1f3e:~# ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_feedback
[INFO] [1714709918.211849627] [minimal_action_client]: Sending goal
[INFO] [1714709918.212104635] [minimal_action_client]: Waiting for result
[INFO] [1714709918.212164000] [minimal_action_client]: Next number in sequence received: 1
[INFO] [1714709919.212808485] [minimal_action_client]: Next number in sequence received: 2
[INFO] [1714709920.212750935] [minimal_action_client]: Next number in sequence received: 3
[INFO] [1714709921.212607189] [minimal_action_client]: Next number in sequence received: 5
[INFO] [1714709922.212611114] [minimal_action_client]: Next number in sequence received: 8
[INFO] [1714709923.212792544] [minimal_action_client]: Next number in sequence received: 13
[INFO] [1714709924.212840838] [minimal_action_client]: Next number in sequence received: 21
[INFO] [1714709925.212954818] [minimal_action_client]: Next number in sequence received: 34
[INFO] [1714709926.212725233] [minimal_action_client]: Next number in sequence received: 55
[INFO] [1714709927.212940610] [minimal_action_client]: result received
[INFO] [1714709927.213027947] [minimal_action_client]: 0
[INFO] [1714709927.213044822] [minimal_action_client]: 1
[INFO] [1714709927.213062440] [minimal_action_client]: 1
[INFO] [1714709927.213077518] [minimal_action_client]: 2
[INFO] [1714709927.213091719] [minimal_action_client]: 3
[INFO] [1714709927.213106753] [minimal_action_client]: 5
[INFO] [1714709927.213144857] [minimal_action_client]: 8
[INFO] [1714709927.213161814] [minimal_action_client]: 13
[INFO] [1714709927.213177762] [minimal_action_client]: 21
[INFO] [1714709927.213193713] [minimal_action_client]: 34
[INFO] [1714709927.213209174] [minimal_action_client]: 55
root@51cdd59e1f3e:~# ros2 run examples_rclcpp_minimal_action_client action_client_member_functions
[INFO] [1714709935.655265723] [minimal_action_client]: Sending goal
[INFO] [1714709935.655719248] [minimal_action_client]: Goal accepted by server, waiting for result
[INFO] [1714709935.655798264] [minimal_action_client]: Next number in sequence received: 1
[INFO] [1714709936.655931117] [minimal_action_client]: Next number in sequence received: 2
[INFO] [1714709937.655923666] [minimal_action_client]: Next number in sequence received: 3
[INFO] [1714709938.655914134] [minimal_action_client]: Next number in sequence received: 5
[INFO] [1714709939.655909595] [minimal_action_client]: Next number in sequence received: 8
[INFO] [1714709940.655926626] [minimal_action_client]: Next number in sequence received: 13
[INFO] [1714709941.655950682] [minimal_action_client]: Next number in sequence received: 21
[INFO] [1714709942.655891613] [minimal_action_client]: Next number in sequence received: 34
[INFO] [1714709943.655913556] [minimal_action_client]: Next number in sequence received: 55
[INFO] [1714709944.655933069] [minimal_action_client]: Result received
[INFO] [1714709944.655952435] [minimal_action_client]: 0
[INFO] [1714709944.655955441] [minimal_action_client]: 1
[INFO] [1714709944.655957556] [minimal_action_client]: 1
[INFO] [1714709944.655959663] [minimal_action_client]: 2
[INFO] [1714709944.655961731] [minimal_action_client]: 3
[INFO] [1714709944.655963679] [minimal_action_client]: 5
[INFO] [1714709944.655967407] [minimal_action_client]: 8
[INFO] [1714709944.655969307] [minimal_action_client]: 13
[INFO] [1714709944.655971774] [minimal_action_client]: 21
[INFO] [1714709944.655973666] [minimal_action_client]: 34
[INFO] [1714709944.655975545] [minimal_action_client]: 55
```

no problem observed, we can close this.

marcoag, 2024-05-03:
Thanks for testing!

closed 2024-05-03 by marcoag