osrf/ros2_test_cases #1248: Executables in `examples_rclcpp_multithreaded_executor`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Check `multithreaded_executor`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_multithreaded_executor multithreaded_executor
```

fujitatomoya, 2024-05-03:
@marcoag i can take care of this.

fujitatomoya, 2024-05-03:
test result.

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclcpp_multithreaded_executor multithreaded_executor
[INFO] [1714709717.368482436] [PublisherNode]:
<<THREAD 2424208470210016361>> Publishing 'Hello World! 0'
[INFO] [1714709717.369090820] [DualThreadedNode]: THREAD 9112217722233392654 => Heard 'Hello World! 0' at 1714709717369085748
[INFO] [1714709717.369124120] [DualThreadedNode]: THREAD 2662725730946419056 => Heard 'Hello World! 0' at 1714709717369119325
[INFO] [1714709717.868373932] [PublisherNode]:
<<THREAD 15880113842283508494>> Publishing 'Hello World! 1'
[INFO] [1714709717.868477504] [DualThreadedNode]: THREAD 15880113842283508494 => Heard 'Hello World! 1' at 1714709717868476734
[INFO] [1714709717.868479532] [DualThreadedNode]: THREAD 14503772475690451546 => Heard 'Hello World! 1' at 1714709717868478459
[INFO] [1714709718.368533722] [PublisherNode]:
<<THREAD 3514649645856480847>> Publishing 'Hello World! 2'
[INFO] [1714709718.368962222] [DualThreadedNode]: THREAD 10875006599289297092 => Heard 'Hello World! 2' at 1714709718368957757
[INFO] [1714709718.369020949] [DualThreadedNode]: THREAD 5972743018955251879 => Heard 'Hello World! 2' at 1714709718369014473
[INFO] [1714709718.868564508] [PublisherNode]:
<<THREAD 5140394814730981491>> Publishing 'Hello World! 3'
[INFO] [1714709718.868908881] [DualThreadedNode]: THREAD 233280648420699347 => Heard 'Hello World! 3' at 1714709718868903764
[INFO] [1714709718.869002222] [DualThreadedNode]: THREAD 2424208470210016361 => Heard 'Hello World! 3' at 1714709718868994925
[INFO] [1714709719.368532356] [PublisherNode]:
<<THREAD 9112217722233392654>> Publishing 'Hello World! 4'
[INFO] [1714709719.368929598] [DualThreadedNode]: THREAD 10744326447835508497 => Heard 'Hello World! 4' at 1714709719368924657
[INFO] [1714709719.369057654] [DualThreadedNode]: THREAD 16254726428639847058 => Heard 'Hello World! 4' at 1714709719369051912
[INFO] [1714709719.868504276] [PublisherNode]:
<<THREAD 15880113842283508494>> Publishing 'Hello World! 5'
[INFO] [1714709719.868834304] [DualThreadedNode]: THREAD 2366937944960823361 => Heard 'Hello World! 5' at 1714709719868828764
[INFO] [1714709719.868938302] [DualThreadedNode]: THREAD 3514649645856480847 => Heard 'Hello World! 5' at 1714709719868932814
[INFO] [1714709720.368458966] [PublisherNode]:
<<THREAD 10875006599289297092>> Publishing 'Hello World! 6'
[INFO] [1714709720.368761981] [DualThreadedNode]: THREAD 10875006599289297092 => Heard 'Hello World! 6' at 1714709720368758189
[INFO] [1714709720.368809589] [DualThreadedNode]: THREAD 18308068040152589952 => Heard 'Hello World! 6' at 1714709720368802938
[INFO] [1714709720.868445150] [PublisherNode]:
<<THREAD 233280648420699347>> Publishing 'Hello World! 7'
[INFO] [1714709720.868701252] [DualThreadedNode]: THREAD 233280648420699347 => Heard 'Hello World! 7' at 1714709720868697305
[INFO] [1714709720.868708438] [DualThreadedNode]: THREAD 2662725730946419056 => Heard 'Hello World! 7' at 1714709720868703911
^C[INFO] [1714709721.305295548] [rclcpp]: signal_handler(signum=2)
```

no problem observed, we can close this.

marcoag, 2024-05-03:
Thanks for testing!

closed 2024-05-03 by marcoag