osrf/ros2_test_cases #1250: Executables in `examples_rclpy_minimal_action_server`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `server_defer`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_action_server server_defer
```
                
    

    
    
            
### Step 2 - Check `server_not_composable`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_action_server server_not_composable
```
                
    

    
    
            
### Step 3 - Check `server`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_action_server server
```
                
    

    
    
            
### Step 4 - Check `server_queue_goals`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_action_server server_queue_goals
```
                
    

    
    
            
### Step 5 - Check `server_single_goal`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_action_server server_single_goal
```

fujitatomoya, 2024-05-03:
@marcoag i can take care of this.

fujitatomoya, 2024-05-03:
test result with `ros2 run examples_rclpy_minimal_action_client client`.

<details>
  <summary>Step 1 - Check server_defer</summary>

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_action_server server_defer
[INFO] [1714709427.018929396] [minimal_action_server]: Received goal request
[INFO] [1714709427.019321667] [minimal_action_server]: Deferring execution...
[INFO] [1714709430.023501670] [minimal_action_server]: Executing goal...
[INFO] [1714709430.024784051] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1])
[INFO] [1714709431.026654290] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2])
[INFO] [1714709432.028619507] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1714709433.030538971] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1714709434.032488983] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1714709435.034422130] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1714709436.036421439] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1714709437.038342600] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1714709438.040310525] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1714709439.043354941] [minimal_action_server]: Returning result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

</details>

<details>
  <summary>Step 2 - Check server_not_composable</summary>

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_action_server server_not_composable
[INFO] [1714709453.039975880] [minimal_action_server]: Executing goal...
[INFO] [1714709453.040211815] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1])
[INFO] [1714709454.041751695] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2])
[INFO] [1714709455.043706773] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1714709456.045563012] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1714709457.047492637] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1714709458.049353596] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1714709459.051267177] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1714709460.053238208] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1714709461.055171024] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1714709462.058155230] [minimal_action_server]: Returning result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

</details>

<details>
  <summary>Step 3 - Check server</summary>

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_action_server server
[INFO] [1714709523.313092574] [minimal_action_server]: Received goal request
[INFO] [1714709523.314054937] [minimal_action_server]: Executing goal...
[INFO] [1714709523.314797542] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1])
[INFO] [1714709524.316508680] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2])
[INFO] [1714709525.317587451] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1714709526.319399717] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1714709527.321342845] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1714709528.323271387] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1714709529.325186414] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1714709530.327039168] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1714709531.328949461] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1714709532.331934152] [minimal_action_server]: Returning result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

</details>

<details>
  <summary>Step 4 - Check server_queue_goals</summary>

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_action_server server_queue_goals
[... 13 lines cut ...]

</details>

<details>
  <summary>Step 5 - Check server_single_goal</summary>

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_action_server server_single_goal
[INFO] [1714709612.093938894] [minimal_action_server]: Received goal request
[INFO] [1714709612.094548569] [minimal_action_server]: Executing goal...
[INFO] [1714709612.095146432] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1])
[INFO] [1714709613.096882524] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2])
[INFO] [1714709614.098895767] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1714709615.100967982] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1714709616.101992392] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[... cut ...]
[INFO] [1714709618.105730666] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1714709619.107855247] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1714709620.108680172] [minimal_action_server]: Publishing feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1714709621.111486646] [minimal_action_server]: Returning result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

</details>

no problem observed, we can close this one.

marcoag, 2024-05-03:
Thanks for checking!

closed 2024-05-03 by marcoag