osrf/ros2_test_cases #1252: Executables in `examples_rclpy_minimal_action_client`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `client_cancel`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_action_client client_cancel
```
                
    

    
    
            
### Step 2 - Check `client_not_composable`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_action_client client_not_composable
```
                
    

    
    
            
### Step 3 - Check `client`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_action_client client
```

fujitatomoya, 2024-05-03:
@marcoag i can take care of this.

fujitatomoya, 2024-05-03:
test result against `ros2 run examples_rclpy_minimal_action_server server`.

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_action_client client_cancel
[INFO] [1714709294.101740687] [minimal_action_client]: Waiting for action server...
[INFO] [1714709294.353279349] [minimal_action_client]: Sending goal request...
[INFO] [1714709294.369116824] [minimal_action_client]: Goal accepted :)
[INFO] [1714709294.369829010] [minimal_action_client]: Received feedback: array('i', [0, 1, 1])
[INFO] [1714709295.373196459] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2])
[INFO] [1714709296.370786107] [minimal_action_client]: Canceling goal
[INFO] [1714709296.379434439] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1714709296.380784477] [minimal_action_client]: Goal successfully canceled
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_action_client client_not_composable
[INFO] [1714709304.259000958] [minimal_action_client]: Waiting for action server...
[INFO] [1714709304.259260015] [minimal_action_client]: Sending goal request...
[INFO] [1714709304.260981542] [minimal_action_client]: Goal accepted :)
[INFO] [1714709304.262076553] [minimal_action_client]: Received feedback: array('i', [0, 1, 1])
[INFO] [1714709305.265554584] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2])
[INFO] [1714709306.267513068] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1714709307.269240734] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1714709308.271380834] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1714709309.273252157] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1714709310.275357360] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1714709311.277270439] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1714709312.280875866] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1714709313.285501330] [minimal_action_client]: Goal succeeded! Result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_action_client client
[INFO] [1714709319.045655133] [minimal_action_client]: Waiting for action server...
[INFO] [1714709319.297310457] [minimal_action_client]: Sending goal request...
[INFO] [1714709319.304488927] [minimal_action_client]: Goal accepted :)
[INFO] [1714709319.310814054] [minimal_action_client]: Received feedback: array('i', [0, 1, 1])
[INFO] [1714709320.312857380] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2])
[INFO] [1714709321.315115673] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1714709322.316758420] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1714709323.318773990] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1714709324.320706161] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1714709325.322676134] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1714709326.324930749] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1714709327.325667559] [minimal_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1714709328.330477450] [minimal_action_client]: Goal succeeded! Result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

i think we can close this.

marcoag, 2024-05-03:
Thanks a lot for testing!

closed 2024-05-03 by marcoag