osrf/ros2_test_cases #1254: Executables in `examples_rclpy_minimal_publisher`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `publisher_local_function`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_publisher publisher_local_function
```
                
    

    
    
            
### Step 2 - Check `publisher_member_function`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_publisher publisher_member_function
```
                
    

    
    
            
### Step 3 - Check `publisher_old_school`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_publisher publisher_old_school
```

fujitatomoya, 2024-05-03:
@marcoag i can take care of this.

fujitatomoya, 2024-05-03:
test result.

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_publisher publisher_local_function
[INFO] [1714700884.462507041] [minimal_publisher]: Publishing: "Hello World: 0"
[INFO] [1714700884.952743856] [minimal_publisher]: Publishing: "Hello World: 1"
[INFO] [1714700885.452702351] [minimal_publisher]: Publishing: "Hello World: 2"
[INFO] [1714700885.952671707] [minimal_publisher]: Publishing: "Hello World: 3"
^CTraceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_minimal_publisher/publisher_local_function", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-publisher==0.19.3', 'console_scripts', 'publisher_local_function')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_minimal_publisher/publisher_local_function.py", line 39, in main
    rclpy.spin(node)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 240, in spin
    executor.spin_once()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 827, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 811, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 685, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_publisher publisher_member_function
[INFO] [1714700901.981832958] [minimal_publisher]: Publishing: "Hello World: 0"
[INFO] [1714700902.471311854] [minimal_publisher]: Publishing: "Hello World: 1"
[INFO] [1714700902.971259807] [minimal_publisher]: Publishing: "Hello World: 2"
[INFO] [1714700903.471259677] [minimal_publisher]: Publishing: "Hello World: 3"
^CTraceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_minimal_publisher/publisher_member_function", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-publisher==0.19.3', 'console_scripts', 'publisher_member_function')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_minimal_publisher/publisher_member_function.py", line 43, in main
    rclpy.spin(minimal_publisher)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 240, in spin
    executor.spin_once()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 827, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 811, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 685, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_publisher publisher_old_school
[INFO] [1714700917.558342518] [minimal_publisher]: Publishing: "Hello World: 0"
[INFO] [1714700918.059980504] [minimal_publisher]: Publishing: "Hello World: 1"
[INFO] [1714700918.561751209] [minimal_publisher]: Publishing: "Hello World: 2"
[INFO] [1714700919.063436760] [minimal_publisher]: Publishing: "Hello World: 3"
[INFO] [1714700919.565215822] [minimal_publisher]: Publishing: "Hello World: 4"
^CTraceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_minimal_publisher/publisher_old_school", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-publisher==0.19.3', 'console_scripts', 'publisher_old_school')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_minimal_publisher/publisher_old_school.py", line 42, in main
    sleep(0.5)  # seconds
    ^^^^^^^^^^
KeyboardInterrupt
[ros2run]: Interrupt
```

same problem https://github.com/ros2/examples/issues/378 can be observed, but application works.
i think we can close this one.

marcoag, 2024-05-03:
Great, thanks!

closed 2024-05-03 by marcoag