osrf/ros2_test_cases #1256: Executables in `examples_rclpy_minimal_subscriber`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `subscriber_lambda`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_subscriber subscriber_lambda
```
                
    

    
    
            
### Step 2 - Check `subscriber_member_function`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_subscriber subscriber_member_function
```
                
    

    
    
            
### Step 3 - Check `subscriber_old_school`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_subscriber subscriber_old_school
```

fujitatomoya, 2024-05-03:
@marcoag i can take care of this.

fujitatomoya, 2024-05-03:
test result.

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_subscriber subscriber_lambda
[INFO] [1714700884.472057801] [minimal_subscriber]: I heard: "Hello World: 0"
[INFO] [1714700884.954768679] [minimal_subscriber]: I heard: "Hello World: 1"
[INFO] [1714700885.454845085] [minimal_subscriber]: I heard: "Hello World: 2"
[INFO] [1714700885.954629035] [minimal_subscriber]: I heard: "Hello World: 3"
^CTraceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_minimal_subscriber/subscriber_lambda", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-subscriber==0.19.3', 'console_scripts', 'subscriber_lambda')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_minimal_subscriber/subscriber_lambda.py", line 29, in main
    rclpy.spin(node)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 240, in spin
    executor.spin_once()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 827, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 811, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 685, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_subscriber subscriber_member_function
[INFO] [1714700901.981827324] [minimal_subscriber]: I heard: "Hello World: 0"
[INFO] [1714700902.472018614] [minimal_subscriber]: I heard: "Hello World: 1"
[INFO] [1714700902.971904731] [minimal_subscriber]: I heard: "Hello World: 2"
[INFO] [1714700903.471900370] [minimal_subscriber]: I heard: "Hello World: 3"
^CTraceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_minimal_subscriber/subscriber_member_function", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-subscriber==0.19.3', 'console_scripts', 'subscriber_member_function')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_minimal_subscriber/subscriber_member_function.py", line 41, in main
    rclpy.spin(minimal_subscriber)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 240, in spin
    executor.spin_once()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 827, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 811, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 685, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_subscriber subscriber_old_school
[INFO] [1714700918.072196545] [minimal_subscriber]: I heard: "Hello World: 1"
[INFO] [1714700918.563724811] [minimal_subscriber]: I heard: "Hello World: 2"
[INFO] [1714700919.065507354] [minimal_subscriber]: I heard: "Hello World: 3"
[INFO] [1714700919.567118805] [minimal_subscriber]: I heard: "Hello World: 4"
^CTraceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_minimal_subscriber/subscriber_old_school", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-subscriber==0.19.3', 'console_scripts', 'subscriber_old_school')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_minimal_subscriber/subscriber_old_school.py", line 38, in main
    rclpy.spin_once(g_node)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 220, in spin_once
    executor.spin_once(timeout_sec=timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 827, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 811, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 685, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt
```

https://github.com/ros2/examples/issues/378 can be observed, but subscription works.
i think we can close this one.

closed 2024-05-03 by marcoag