osrf/ros2_test_cases #1258: Executables in `examples_rclpy_minimal_client`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `client`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_client client
```
                
    

    
    
            
### Step 2 - Check `client_async`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_client client_async
```
                
    

    
    
            
### Step 3 - Check `client_async_member_function`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_client client_async_member_function
```
                
    

    
    
            
### Step 4 - Check `client_async_callback`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_client client_async_callback
```

fujitatomoya, 2024-05-03:
@marcoag i will take care of this.

fujitatomoya, 2024-05-03:
test result.

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_client client
[INFO] [1714698877.287270928] [minimal_client]: Result of add_two_ints: for 41 + 1 = 42
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_client client_async
[INFO] [1714698883.973607972] [minimal_client_async]: Result of add_two_ints: for 41 + 1 = 42
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_client client_async_member_function
[INFO] [1714698904.535175007] [minimal_client_async]: Result of add_two_ints: for 41 + 1 = 42
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_client client_async_callback
[INFO] [1714698911.111156869] [minimal_client]: Result of add_two_ints: for 41 + 1 = 42
```

no problem observed, we can close this.

closed 2024-05-03 by marcoag