osrf/ros2_test_cases #1259: Executables in `examples_rclpy_minimal_client`

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Process

    
            
### Step 1 - Check `client`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_client client
```
                
    

    
    
            
### Step 2 - Check `client_async`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_client client_async
```
                
    

    
    
            
### Step 3 - Check `client_async_member_function`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_client client_async_member_function
```
                
    

    
    
            
### Step 4 - Check `client_async_callback`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_client client_async_callback
```

mikaelarguedas, 2024-05-04:
Tested along https://github.com/osrf/ros2_test_cases/issues/1261

## Step 1 - Check client

Test passed ✔️
<details>

```
root@d8ff93e471d7:/opt/ros/jazzyros2 run examples_rclpy_minimal_client clientnt
[INFO] [1714855663.633049946] [minimal_client]: Result of add_two_ints: for 41 + 1 = 42
```

</details>

## Step 2 - Check client_async
Test passed ✔️
<details>

```
[INFO] [1714855682.005823708] [minimal_client_async]: Result of add_two_ints: for 41 + 1 = 42
```

</details>

## Step 3 - Check client_async_member_function
Test passed ✔️
<details>

```
[INFO] [1714855686.396758637] [minimal_client_async]: Result of add_two_ints: for 41 + 1 = 42
```

</details>

## Step 4 - Check client_async_callback
Test failed :x: 

The server processes the request but the client never processes the response and hangs forever

<details>

```
root@d8ff93e471d7:/opt/ros/jazzy# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@d8ff93e471d7:/opt/ros/jazzy# ros2 run examples_rclpy_minimal_client client_async_callback
^CTraceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_minimal_client/client_async_callback", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-client==0.19.3', 'console_scripts', 'client_async_callback')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_minimal_client/client_async_callback.py", line 60, in main
    rclpy.spin_once(node)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 220, in spin_once
    executor.spin_once(timeout_sec=timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 827, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 811, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 739, in _wait_for_ready_callbacks
    if client.handle.pointer in clients_ready:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt
[ros2run]: Interrupt
```

</details>

Note that with rmw_fastrtps_cpp it works:
```
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
root@d8ff93e471d7:/opt/ros/jazzy# ros2 run examples_rclpy_minimal_client client_async_callback
[INFO] [1714856133.085582940] [minimal_client]: Result of add_two_ints: for 41 + 1 = 42
```

samuel-ongzx, 2024-05-14:
Just tested this and second @mikaelarguedas findings. `FastDDS` works with Step 4, but `CycloneDDS` does not. Like what Mike mentioned, the request is processed by the service, but the client does not terminate. when using `CycloneDDS`. 

Setup: Docker `ghcr.io/sloretz/ros-testing:jazzy-desktop` on AMD64, installed `CycloneDDS` from Debian package.

clalancette, 2024-05-16:
See https://github.com/ros2/examples/pull/382, which should fix the issue here.

clalancette, 2024-05-16:
With that PR open, I'm going to go ahead and close this.  Thank you for testing!

closed 2024-05-16 by clalancette