osrf/ros2_test_cases #1260: Executables in `examples_rclpy_minimal_service`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `service`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_service service
```
                
    

    
    
            
### Step 2 - Check `service_member_function`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_service service_member_function
```

fujitatomoya, 2024-05-03:
@marcoag i can take care of this.

fujitatomoya, 2024-05-03:
- Step 1

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_service service
[INFO] [1714698877.276901703] [minimal_service]: Incoming request
a: 41 b: 1
[INFO] [1714698883.966641985] [minimal_service]: Incoming request
a: 41 b: 1
[INFO] [1714698904.528264472] [minimal_service]: Incoming request
a: 41 b: 1
[INFO] [1714698911.101914394] [minimal_service]: Incoming request
a: 41 b: 1
```

- Step 2

```bash
root@51cdd59e1f3e:~# ros2 run examples_rclpy_minimal_service service_member_function
[INFO] [1714698951.435342024] [minimal_service]: Incoming request
a: 41 b: 1
[INFO] [1714698955.953502812] [minimal_service]: Incoming request
a: 41 b: 1
[INFO] [1714698959.289201885] [minimal_service]: Incoming request
a: 41 b: 1
[INFO] [1714698963.647180164] [minimal_service]: Incoming request
a: 41 b: 1
```

there is one thing, when signal is sent to interrupt, the following stack trace comes up.
this can be observed with `rolling` so i created issue https://github.com/ros2/examples/issues/378 to track.

```bash
^CTraceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_minimal_service/service_member_function", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-service==0.19.3', 'console_scripts', 'service_member_function')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_minimal_service/service_member_function.py", line 39, in main
    rclpy.spin(minimal_service)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 240, in spin
    executor.spin_once()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 827, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 811, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 685, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt
```

fujitatomoya, 2024-05-03:
I think we can close this.

marcoag, 2024-05-03:
Perfect, thanks!

closed 2024-05-03 by marcoag