osrf/ros2_test_cases #1261: Executables in `examples_rclpy_minimal_service`

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Process

    
            
### Step 1 - Check `service`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_service service
```
                
    

    
    
            
### Step 2 - Check `service_member_function`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_minimal_service service_member_function
```

mikaelarguedas, 2024-05-04:
Tested along https://github.com/osrf/ros2_test_cases/issues/1259

Tests passed :heavy_check_mark: 

## Step 1 - Check service

<details>

```
[INFO] [1714855663.622473331] [minimal_service]: Incoming request
a: 41 b: 1
```

</details>

## Step 2 - Check service_member_function

<details>

```
root@d8ff93e471d7:/# ros2 run examples_rclpy_minimal_service service_member_function
[INFO] [1714855808.178485526] [minimal_service]: Incoming request
a: 41 b: 1
```

</details>

---

:warning:  Traceback on KeyboardInterrupt

```
Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_minimal_service/service_member_function", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-service==0.19.3', 'console_scripts', 'service_member_function')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_minimal_service/service_member_function.py", line 39, in main
    rclpy.spin(minimal_service)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 240, in spin
    executor.spin_once()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 827, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 811, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 685, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt
```

clalancette, 2024-05-07:
The `KeyboardInterrupt` problem will be fixed by https://github.com/ros2/examples/pull/379 , so closing this out.  Thanks for testing!

closed 2024-05-07 by clalancette