osrf/ros2_test_cases #1263: Executables in `examples_rclpy_executors`

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Process

    
            
### Step 1 - Check `callback_group`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_executors callback_group
```
                
    

    
    
            
### Step 2 - Check `composed`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_executors composed
```
                
    

    
    
            
### Step 3 - Check `custom_callback_group`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_executors custom_callback_group
```
                
    

    
    
            
### Step 4 - Check `custom_executor`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_executors custom_executor
```
                
    

    
    
            
### Step 5 - Check `listener`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_executors listener
```
                
    

    
    
            
### Step 6 - Check `talker`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclpy_executors talker
```

mikaelarguedas, 2024-05-04:
Tests passed ✔️

## Step 1 - Check callback_group

Test passed ✔️

<details>

```
root@d8ff93e471d7:/opt/ros/jazzy# ros2 run examples_rclpy_executors callback_group
[INFO] [1714854772.690380990] [double_talker]: Publishing: "Hello World: 0"
[INFO] [1714854772.691833370] [listener]: I heard: "Hello World: 0"
[INFO] [1714854773.179725720] [double_talker]: Publishing: "Hello World: 1"
[INFO] [1714854773.181039059] [double_talker]: Publishing: "Hello World: 2"
[INFO] [1714854773.181367779] [listener]: I heard: "Hello World: 1"
[INFO] [1714854773.182036230] [listener]: I heard: "Hello World: 2"
[INFO] [1714854773.679438728] [double_talker]: Publishing: "Hello World: 3"
[INFO] [1714854773.680442159] [listener]: I heard: "Hello World: 3"
[INFO] [1714854774.180643576] [double_talker]: Publishing: "Hello World: 4"
[INFO] [1714854774.181946294] [double_talker]: Publishing: "Hello World: 5"
[INFO] [1714854774.182286870] [listener]: I heard: "Hello World: 4"
[INFO] [1714854774.182934428] [listener]: I heard: "Hello World: 5"
```

</details>

## Step 2 - Check composed

Test passed ✔️

<details>

```
root@d8ff93e471d7:/opt/ros/jazzy# ros2 run examples_rclpy_executors composed
[INFO] [1714854980.905303534] [talker]: Publishing: "Hello World: 0"
[INFO] [1714854980.906060623] [listener]: I heard: "Hello World: 0"
[INFO] [1714854981.895956002] [talker]: Publishing: "Hello World: 1"
[INFO] [1714854981.897665414] [listener]: I heard: "Hello World: 1"
[INFO] [1714854982.895129943] [talker]: Publishing: "Hello World: 2"
[INFO] [1714854982.895835718] [listener]: I heard: "Hello World: 2"
[INFO] [1714854983.895062353] [talker]: Publishing: "Hello World: 3"
[INFO] [1714854983.895775443] [listener]: I heard: "Hello World: 3"
[INFO] [1714854984.895028327] [talker]: Publishing: "Hello World: 4"
[INFO] [1714854984.895780370] [listener]: I heard: "Hello World: 4"
[INFO] [1714854985.895040023] [talker]: Publishing: "Hello World: 5"
[INFO] [1714854985.895830531] [listener]: I heard: "Hello World: 5"
```

</details>

## Step 3 - Check custom_callback_group

Test passed ✔️

<details>

```
root@d8ff93e471d7:/opt/ros/jazzy# ros2 run examples_rclpy_executors custom_callback_group
[INFO] [1714855031.671087026] [intermittent_talker]: Publishing: "Hello World: 0"
[INFO] [1714855031.760959487] [intermittent_talker]: Publishing: "Hello World: 1"
[INFO] [1714855031.860960666] [intermittent_talker]: Publishing: "Hello World: 2"
[INFO] [1714855031.960925493] [intermittent_talker]: Publishing: "Hello World: 3"
[INFO] [1714855032.060975176] [intermittent_talker]: Publishing: "Hello World: 4"
[INFO] [1714855032.161652560] [intermittent_talker]: Publishing: "Hello World: 5"
[INFO] [1714855032.260918398] [intermittent_talker]: Publishing: "Hello World: 6"
[INFO] [1714855032.361774195] [intermittent_talker]: Publishing: "Hello World: 7"
[INFO] [1714855032.460927184] [intermittent_talker]: Publishing: "Hello World: 8"
[INFO] [1714855032.560958518] [intermittent_talker]: Publishing: "Hello World: 9"
[INFO] [1714855032.660874076] [intermittent_talker]: Publishing: "Hello World: 10"
[INFO] [1714855032.760886397] [intermittent_talker]: Publishing: "Hello World: 11"
[... 93 lines cut ...]
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 685, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/examples_rclpy_executors/callback_group", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-executors==0.19.3', 'console_scripts', 'callback_group')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/examples_rclpy_executors/callback_group.py", line 68, in main
    rclpy.shutdown()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 130, in shutdown
    _shutdown(context=context)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/utilities.py", line 82, in shutdown
    context.shutdown()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/context.py", line 129, in shutdown
    self.__context.shutdown()
rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at ./src/rcl/init.c:333
[ros2run]: Process exited with failure 1
```

clalancette, 2024-05-07:
Both of those issues will be fixed by https://github.com/ros2/examples/pull/379 , so closing this out.  Thanks for testing!

closed 2024-05-07 by clalancette