osrf/ros2_test_cases #1264: Executables in `pendulum_control`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Step 1 - Check `pendulum_demo`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run pendulum_control pendulum_demo
```
                
    

    
    
            
### Step 2 - Check `pendulum_teleop`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run pendulum_control pendulum_teleop
```
                
    

    
    
            
### Step 3 - Check `pendulum_launch.bash`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 launch pendulum_control pendulum_launch.bash
```
                
    

    
    
            
### Step 4 - Check `pendulum_logger`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run pendulum_control pendulum_logger
```

fujitatomoya, 2024-05-03:
@marcoag you can sign me up for this.

fujitatomoya, 2024-05-03:
Only test 2 cannot start the test script since there is no such file....

```bash
root@51cdd59e1f3e:~# ros2 launch pendulum_control pendulum_launch.bash
file 'pendulum_launch.bash' was not found in the share directory of package 'pendulum_control' which is at '/opt/ros/jazzy/share/pendulum_control'
```

other tests are no problem.

<details>
  <summary>Step 1 - Check pendulum_demo</summary>

```bash
root@51cdd59e1f3e:~# ros2 run pendulum_control pendulum_demo
Initial major pagefaults: 11
Initial minor pagefaults: 2133285
No results filename given, not writing results
rttest statistics:
  - Minor pagefaults: 0
  - Major pagefaults: 0
  Latency (time after deadline was missed):
    - Min: 7016 ns
    - Max: 192573 ns
    - Mean: 21091.233000 ns
    - Standard deviation: 29905.110308

PendulumMotor received 497 messages
PendulumController received 499 messages
```

</details>

<details>
  <summary>Step 2 - Check pendulum_teleop</summary>

```bash
root@51cdd59e1f3e:~# ros2 run pendulum_control pendulum_teleop
Command argument not specified. Setting command to 90 degrees (PI/2 radians).
Teleop message published.
Teleop node exited.
```

</details>

<details>
  <summary>Step 4 - Check pendulum_logger</summary>

```bash
root@51cdd59e1f3e:~# ros2 run pendulum_control pendulum_logger
Logger node initialized.
Commanded motor angle: 1.570796
Actual motor angle: 0.000000
Current latency: 24186 ns
Mean latency: 24186.000000 ns
Min latency: 24186 ns
Max latency: 24186 ns
Minor pagefaults during execution: 0
Major pagefaults during execution: 0
...
```

</details>

closed 2024-05-03 by marcoag

clalancette, 2024-05-03:
I opened up https://github.com/osrf/ros2_test_cases/pull/1546 to remove pendulum_launch.bash from the test case.