osrf/ros2_test_cases #1267: Executables in `tlsf_cpp`

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Process

    
            
### Check `tlsf_allocator_example`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run tlsf_cpp tlsf_allocator_example
```

knmcguire, 2024-05-06:
You can assign me for this.

knmcguire, 2024-05-06:
Tried it out with jazzy docker on an AMD64 ubuntu machine.

```
root@pop-os:/# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@pop-os:/# ros2 run tlsf_cpp tlsf_allocator_example
Intra-process pipeline is OFF.
root@pop-os:/# ros2 run tlsf_cpp tlsf_allocator_example intra
Intra-process pipeline is ON.
```
**seems to work**

clalancette, 2024-05-06:
Thanks for testing!

closed 2024-05-06 by clalancette