osrf/ros2_test_cases #1268: Executables in `rttest`

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Check `rttest_plot`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 run rttest rttest_plot
```

Kartik9250, 2024-05-03:
I would like to work on this

Kartik9250, 2024-05-03:
The `ros2 run rttest rttest_plot` give out the output given below:
```usage: rttest_plot [-h] [-s] [-o OUTFILE] filename
rttest_plot: error: the following arguments are required: filename
[ros2run]: Process exited with failure 2
```
What file am I supposed to provide it.

clalancette, 2024-05-06:
> What file am I supposed to provide it.

Hm, good question.  I'll take a look and see if I can figure it out.

clalancette, 2024-05-06:
All right, I figured it out.  I opened up https://github.com/osrf/ros2_test_cases/pull/1553 to update the test case, and I manually modified this test case to show the proper instructions.  @Kartik9250 you should be able to follow and test this one now.

clalancette, 2024-05-14:
Since I think this one is solved, I'm going to go ahead and close it.  Thanks for testing!

closed 2024-05-14 by clalancette