osrf/ros2_test_cases #1274: Executables in gz vendor packages

marcoag, 2024-04-30:
You will need to install the `desktop-full` or `simulation` variants before running this test.

## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Check `gz sim -v4 shapes.sdf`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
gz sim -v4 shapes.sdf
```

Ayush1285, 2024-05-02:
This works fine as expected.
[image]

But if we go to the tutorial([link](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Simulators/Gazebo/Gazebo.html)) provided in Documentation of Jazzy. It still mentions the "ign gazebo ..." command. It is not supported in Gazebo Harmonic(default pairing with Jazzy).

clalancette, 2024-05-02:
> It is not supported in Gazebo Harmonic(default pairing with Jazzy). "ign gazebo" is supported in Gazebo Fortress I think.

Yeah, good point.  We are just waiting for https://github.com/ros2/ros2_documentation/pull/4311 to be finished and land, and then that will be fixed.  I'll leave this open until then.

clalancette, 2024-05-06:
We are close to landing that PR, so closing this out for now.  Thanks for testing!

closed 2024-05-06 by clalancette