osrf/ros2_test_cases #1279: Executables in `rosbag2`

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Process

    
            
### Step 1 - Check `ros2 bag record -a`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 bag record -a
```
                
    

    
    
            
### Step 2 - Check `ros2 bag record <topic1> <topic2> <topic3>`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 bag record <topic1> <topic2> <topic3>
```
                
    

    
    
            
### Step 3 - Check `ros2 bag info <bag_file>`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 bag info <bag_file>
```
                
    

    
    
            
### Step 4 - Check `ros2 bag play <bag_file>`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 bag play <bag_file>
```

mikaelarguedas, 2024-05-05:
messages generated using `ros2 launch composition composition_demo_launch.py`

:heavy_check_mark: All tests passed

Unclear how to test for service recording though :thinking: 

## Step 1 - Check ros2 bag record -a

<details>
```
root@f3de56946821:/# ros2 bag record -a
[INFO] [1714891489.589660092] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1714891489.593873667] [rosbag2_recorder]: Listening for topics...
[INFO] [1714891489.593882197] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1714891489.595157502] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1714891489.595963738] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[INFO] [1714891489.596498248] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1714891489.597413603] [rosbag2_recorder]: Subscribed to topic '/chatter'
[INFO] [1714891489.597466972] [rosbag2_recorder]: Recording...
[INFO] [1714891498.555743427] [rosbag2_recorder]: Pausing recording.
[INFO] [1714891498.555893809] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
[INFO] [1714891498.560370741] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1714891498.560397882] [rosbag2_recorder]: Recording stopped
[INFO] [1714891498.643591681] [rclcpp]: signal_handler(signum=2)
```

</details>

## Step 2 - Check ros2 bag record <topic1> <topic2> <topic3>

<details>

```
root@f3de56946821:/# ros2 bag record /chatter /rosout
[INFO] [1714891513.415839204] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1714891513.418478347] [rosbag2_recorder]: Listening for topics...
[INFO] [1714891513.418486427] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1714891513.419399201] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1714891513.420227834] [rosbag2_recorder]: Subscribed to topic '/chatter'
[INFO] [1714891513.420272604] [rosbag2_recorder]: Recording...
[INFO] [1714891513.420371607] [rosbag2_recorder]: All requested topics are subscribed. Stopping discovery...
[INFO] [1714891520.683906311] [rosbag2_recorder]: Pausing recording.
[INFO] [1714891523.407586748] [rosbag2_recorder]: Resuming recording.
[INFO] [1714891528.323403077] [rosbag2_recorder]: Pausing recording.
[INFO] [1714891528.323601491] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
[INFO] [1714891528.328475719] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1714891528.328502485] [rosbag2_recorder]: Recording stopped
[INFO] [1714891528.412033936] [rclcpp]: signal_handler(signum=2)
```

</details>

## Step 3 - Check ros2 bag info <bag_file>

<details>

```
root@f3de56946821:/# ros2 bag info rosbag2_2024_05_05-06_44_49

Files:             rosbag2_2024_05_05-06_44_49_0.mcap
Bag size:          32.7 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          8.558s
Start:             May  5 2024 06:44:49.595 (1714891489.595)
End:               May  5 2024 06:44:58.153 (1714891498.153)
Messages:          45
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 9 | Serialization Format: cdr
                   Topic: /events/write_split | Type: rosbag2_interfaces/msg/WriteSplitEvent | Count: 0 | Serialization Format: cdr
                   Topic: /parameter_events | Type: rcl_interfaces/msg/ParameterEvent | Count: 0 | Serialization Format: cdr
                   Topic: /rosout | Type: rcl_interfaces/msg/Log | Count: 36 | Serialization Format: cdr
Service:           0
Service information: 
root@f3de56946821:/# ros2 bag info rosbag2_2024_05_05-06_45_13

Files:             rosbag2_2024_05_05-06_45_13_0.mcap
Bag size:          23.8 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          14.733s
Start:             May  5 2024 06:45:13.419 (1714891513.419)
End:               May  5 2024 06:45:28.153 (1714891528.153)
Messages:          57
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 12 | Serialization Format: cdr
                   Topic: /rosout | Type: rcl_interfaces/msg/Log | Count: 45 | Serialization Format: cdr
Service:           0
Service information: 
```

</details>

## Step 4 - Check ros2 bag play <bag_file>

<details>

```
root@f3de56946821:/# ros2 bag play rosbag2_2024_05_05-06_45_13
[INFO] [1714891584.327150906] [rosbag2_player]: Set rate to 1
[INFO] [1714891584.330011716] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1714891584.330026789] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1714891584.330030663] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1714891584.330034300] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1714891584.330037388] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
[INFO] [1714891584.330177807] [rosbag2_player]: Playback until timestamp: -1
```

```
root@f3de56946821:/# ros2 topic echo /chatter
data: 'Hello World: 71'
---
data: 'Hello World: 72'
---
```
</details>

mikaelarguedas, 2024-05-06:
I attempted to provide some information in step 3, not sure where to dig in after as I dont have a source build ATM but it looks like a timestamping issue.

clalancette, 2024-05-06:
> I attempted to provide some information in step 3, not sure where to dig in after as I dont have a source build ATM but it looks like a timestamping issue.

Thanks!

clalancette, 2024-05-06:
@MichaelOrlov this seems to be a severe issue with rosbag2 and CycloneDDS.  Can you take a look (I'll also open an issue over at https://github.com/ros2/rosbag2).

MichaelOrlov, 2024-05-07:
@clalancette @marcoag See RCA in the https://github.com/ros2/rosbag2/issues/1638#issuecomment-2097426872 and workaround in the https://github.com/ros2/rosbag2/pull/1641

closed 2024-06-26 by marcoag