osrf/ros2_test_cases #1281: Check action Cli

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Process

    
            
### Step 1 - Check info
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
```
                
```bash
# StdIn - terminal 2
ros2 action info /fibonacci
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
Action: /fibonacci
Action clients: 0
Action servers: 1
  /minimal_action_server

```
                
    
            
### Step 2 - Check list
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
```
                
```bash
# StdIn - terminal 2
ros2 action list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/fibonacci
```
                
    
            
### Step 3 - Check send_goal
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
```
                
```bash
# StdIn - terminal 2
ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci order:\ 2
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
Waiting for an action server to become available...
Sending goal:
    order: 2

Goal accepted with ID: 0110dbd59e804374bff8e4900c3201e8

Result:
    sequence:
- 0
- 1
- 1

Goal finished with status: SUCCEEDED

```

pradyum, 2024-05-04:
## Environment

```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.8.0-31-generic-x86_64-with-glibc2.39
release          : 6.8.0-31-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

## Result

I get the expected results for the checks mentioned above. This test is passed.

mikaelarguedas, 2024-05-04:
Same here: I got the expecter result :+1:

closed 2024-05-06 by ahcorde