osrf/ros2_test_cases #1283: Check ROS 2 bag CLI

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Process

    
            
### Step 1 - Check record
            
        

    
    

    
        
            
#### Actions
            
        
        
You should stop the talker once the recording is stopped.
                
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp talker
```
                
Run for a few seconds and then hit Ctrl-C to stop the bag recording.
                
```bash
# StdIn - terminal 2
ros2 bag record /chatter -o my-bag
```
                
    

    
        
#### Expected Result
        
        
You should see a folder named my-bag in the current directory
                
    
            
### Step 2 - Check info
            
        

    
    

    
        
            
#### Actions
            
        
        
Continuing from the previous check.
                
```bash
# StdIn - terminal 1
ros2 bag info my-bag
```
                
    

    
        
#### Expected Results
        
        
Something like the following
                
```bash
# StdOut - terminal 1
Files:             my-bag_0.mcap
Bag size:          3.0 KiB
Storage id:        mcap
Duration:          3.999s
Start:             Apr  2 2023 13:45:47.834 (1680414347.834)
End:               Apr  2 2023 13:45:51.834 (1680414351.834)
Messages:          5
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 5 | Serialization Format: cdr

```
                
    
            
### Step 3 - Check play
            
        

    
    

    
        
            
#### Actions
            
        
        
Continuing from the previous two checks. Make sure that the talker is stopped.
                
The `-l` option loops the bag. You should kill this process eventually with Ctrl-C.
                
```bash
# StdIn - terminal 1
ros2 bag play -l my-bag
```
                
```bash
# StdIn - terminal 2
ros2 topic echo /chatter
```
                
    

    
        
#### Expected Results
        
        
Something like the following.
                
```bash
# StdOut - terminal 2
---
data: 'Hello World: 77'
---
data: 'Hello World: 78'
---
data: 'Hello World: 75'
---
data: 'Hello World: 76'
---
data: 'Hello World: 77'
---

```

pradyum, 2024-05-04:
## Environment

```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.8.0-31-generic-x86_64-with-glibc2.39
release          : 6.8.0-31-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```

## Result

**ros2 bag info**

```
pc@pc-Standard-PC-i440FX-PIIX-1996:~$ ros2 bag info my-bag

Files:             my-bag_0.mcap
Bag size:          7.9 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          0.0s
Start:             Jan  1 1970 05:30:00.0 (0.0)
End:               Jan  1 1970 05:30:00.0 (0.0)
Messages:          44
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 44 | Serialization Format: cdr
Service:           0
Service information:
```

The duration of recording is 0.0s and there is problem with start and end time. I Couldn't able to see messages when the bag is played. This test is failed.

samuel-ongzx, 2024-05-15:
Thanks @pradyum for testing. I just tested this and found no issues.

## Setup
OS: Docker - `ghcr.io/sloretz/ros-testing:jazzy-desktop`
Chip set: AMD64
Build type: Debian (preinstalled in docker)
DDS: CycloneDDS (installed with `apt-get`)

## Process
- [x] Step 1: Check record (`ok`)
- [x] Step 2: Check info (`ok`)
- [x] Step 3: Check play (`ok`)

## Verification
Commands: `record`, `info` and `play` are `ok`!
[image]

Starting the talker node
[image]

Echoing the topic after replaying from bag
[image]

samuel-ongzx, 2024-05-15:
Seems like there might be some issue with CycloneDDS & rosbag2, happy to have someone test again if needed. 
Topics referenced: 
1. https://github.com/osrf/ros2_test_cases/issues/1279
2. https://github.com/ros2/rosbag2/issues/1638

I have tested the steps to reproduce the error in https://github.com/ros2/rosbag2/issues/1638 and found no issue, timing looks correct:
[image]

clalancette, 2024-05-16:
We fixed the issue with rosbag2 and cyclonedds in https://github.com/ros2/rmw_cyclonedds/pull/491, which I believe should be released into Jazzy now.  So going ahead and closing this one, thanks for testing!

closed 2024-05-16 by clalancette