osrf/ros2_test_cases #1309: Check ROS 2 service CLI

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Process

    
            
### Step 1 - Check call
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp add_two_ints_server
```
                
```bash
# StdIn - terminal 2
ros2 service call /add_two_ints example_interfaces/srv/AddTwoInts "{a: 1, b: 2}"

```
                
    

    
        
#### Expected Results
        
        
```bash
# StdOut - terminal 1
[INFO] [1649123971.736926946] [add_two_ints_server]: Incoming request
a: 1 b: 2

```
                
```bash
# StdOut - terminal 2
requester: making request: example_interfaces.srv.AddTwoInts_Request(a=1, b=2)

response:
example_interfaces.srv.AddTwoInts_Response(sum=3)

```
                
    
            
### Step 2 - Check find
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp add_two_ints_server
```
                
```bash
# StdIn - terminal 2
ros2 service find example_interfaces/srv/AddTwoInts
```
                
    

    
        
#### Expected Results
        
        
```bash
# StdOut - terminal 2
/add_two_ints
```
                
Trying to find another service message that is not being used will return no text to stdout; for example `ros2 service find example_interfaces/srv/SetBool` returns nothing.
                
    
            
### Step 3 - Check list
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp add_two_ints_server
```
                
```bash
# StdIn - terminal 2
ros2 service list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically

```
                
    
            
### Step 4 - Check type
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp add_two_ints_server
```
                
```bash
# StdIn - terminal 2
ros2 service type /add_two_ints
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
example_interfaces/srv/AddTwoInts
```

hwoithe, 2024-05-02:
I followed the instructions above and **almost** everything worked as expected.

The only slight difference is in the output of step 3.
My output includes the line:
```/add_two_ints_server/get_type_description```

Step 1:
[image]

Step 2:
[image]

Step 3:
[image]

Step 4:
[image]

Here is my system's information:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04 LTS
Release:    24.04
Codename:   noble

$ glxinfo | grep -i renderer
OpenGL renderer string: NVIDIA GeForce RTX 3060 Ti/PCIe/SSE2
```

clalancette, 2024-05-06:
Good call on the missing service.  I've opened up https://github.com/osrf/ros2_test_cases/pull/1551 to fix that for the tests next year.

With that, I'm going to close this.  Thank you for testing!

closed 2024-05-06 by clalancette