osrf/ros2_test_cases #1311: Check ROS 2 topic CLI

marcoag, 2024-04-30:
## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: CycloneDDS

## Process

    
            
### Step 1 - Check bw
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic bw /joint_states
```
                
    

    
        
#### Expected Results
        
        
Something like the following.
                
```bash
# StdOut - terminal 2
Subscribed to [/joint_states]
5.04 KB/s from 50 messages
  Message size mean: 0.10 KB min: 0.10 KB max: 0.10 KB
5.02 KB/s from 100 messages
  Message size mean: 0.10 KB min: 0.10 KB max: 0.10 KB
...

```
                
    
            
### Step 2 - Check delay
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic delay /joint_states
```
                
    

    
        
#### Expected Results
        
        
Something like the following.
                
```bash
# StdOut - terminal 2
average delay: 0.001
  min: 0.000s max: 0.001s std dev: 0.00012s window: 47
average delay: 0.001
  min: 0.000s max: 0.002s std dev: 0.00021s window: 97
...

```
                
    
            
### Step 3 - Check echo
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic echo /joint_states
```
                
    

    
        
#### Expected Results
        
        
Something like the following.
                
```bash
# StdOut - terminal 2
---
header:
  stamp:
    sec: 1649125169
    nanosec: 961952254
  frame_id: ''
name:
- single_rrbot_joint1
- single_rrbot_joint2
position:
- -0.9985924716759744
- -0.9985924716759744
velocity: []
effort: []
---
...

```
                
    
            
### Step 4 - Check find
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic find sensor_msgs/msg/JointState
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/joint_states
```
                
    
            
### Step 5 - Check hz
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic hz /joint_states
```
                
    

    
        
#### Expected Results
        
        
Something like the following.
                
```bash
# StdOut - terminal 2
average rate: 50.252
  min: 0.000s max: 0.043s std dev: 0.00451s window: 52
average rate: 50.125
  min: 0.000s max: 0.043s std dev: 0.00492s window: 102
...

```
                
    
            
### Step 6 - Check info
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic info /joint_states
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
Type: sensor_msgs/msg/JointState
Publisher count: 1
Subscription count: 1

```
                
    
            
### Step 7 - Check list
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/joint_states
/map
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static

```
                
    
            
### Step 8 - Check type
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic type /joint_states
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
sensor_msgs/msg/JointState
```
                
    
            
### Step 9 - Check pub
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp listener
```
                
```bash
# StdIn - terminal 2
ros2 topic pub /chatter std_msgs/msg/String "{data: 'hi'}"

```
                
    

    
        
#### Expected Results
        
        
Something like the following.
                
```bash
# StdOut - terminal 1
[INFO] [1649125551.479505602] [listener]: I heard: [hi]
[INFO] [1649125552.480104544] [listener]: I heard: [hi]
...

```
                
Something like the following.
                
```bash
# StdOut - terminal 2
publisher: beginning loop
publishing #1: std_msgs.msg.String(data='hi')

publishing #2: std_msgs.msg.String(data='hi')
...

```

slowrunner, 2024-05-03:
I added cyclonedds:
```
sudo apt install ros-jazzy-rmw-cyclonedds-cpp

export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp 

and see 
ubuntu      3990    2178  0 07:50 pts/1    00:00:00 /usr/bin/python3 -c from ros2cli.daemon.daemonize import main; main() --name ros2-daemon --ros-domain-id 0 --rmw-implementation rmw_cyclonedds_cpp

```

- Step 1 Check BW:  Passed
```
~/nobledesk$ ros2 topic bw /joint_states
WARNING: topic [/joint_states] does not appear to be published yet
Subscribed to [/joint_states]
5.03 KB/s from 50 messages
	Message size mean: 0.10 KB min: 0.10 KB max: 0.10 KB
5.01 KB/s from 100 messages
	Message size mean: 0.10 KB min: 0.10 KB max: 0.10 KB

```
- Step 2  Check Delay:  Passed  max: 0.001s  std dev: 0.00012s
- Step 3  Check Echo:  Passed
- Step 4 Check Find:  Passed
- Step 5 Check Hz: Passed
```
average rate: 50.001
	min: 0.019s max: 0.021s std dev: 0.00019s window: 304

```
- Step 6 Check Info: Passed
- Step 7 Check List: Passed
- Step 8 Check Type: Passed
- Step 9 Check Pub: Passed

### All steps successfully passed using rmw_cyclonedds_cpp.

clalancette, 2024-05-03:
Thanks for testing!

closed 2024-05-03 by clalancette