osrf/ros2_test_cases #1312: Service introspection cpp

marcoag, 2024-04-30:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Links

* [demo_nodes_cpp](https://github.com/ros2/demos/tree/jazzy/demo_nodes_cpp)

## Process

    
            
### Step 1 - Run the client and server
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp introspect_services_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic list --include-hidden-topics
```
                
    

    
        
#### Expected Results
        
        
```bash
# StdOut - terminal 1
[introspection_service-1] a: 2 b: 3
[introspection_client-2] [INFO] [1680585324.964598437] [introspection_client]: Result of add_two_ints: 5

```
                
```bash
# StdOut - terminal 2
/parameter_events
/rosout

```
                
    
            
### Step 2 - Configure service to send introspection metadata
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp introspect_services_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 param set /introspection_service service_configure_introspection metadata
```
                
```bash
# StdIn - terminal 2
ros2 topic echo /add_two_ints/_service_event
```
                
    

    
        
#### Expected Results
        
        
```bash
# StdOut - terminal 1
[introspection_service-1] a: 2 b: 3
[introspection_client-2] [INFO] [1680585324.964598437] [introspection_client]: Result of add_two_ints: 5

```
                
```bash
# StdOut - terminal 2
info:
  event_type: 1
  stamp:
    sec: 1680586531
    nanosec: 726539263
  client_gid:
  - 83
  - 143
  - 248
  - 38
  - 47
  - 231
  - 185
  - 230
  - 98
  - 17
  - 45
  - 187
  - 128
  - 124
  - 253
  - 156
  sequence_number: 46
request: []
response: []
---
info:
  event_type: 2
  stamp:
    sec: 1680586531
    nanosec: 726695003
  client_gid:
  - 83
  - 143
  - 248
  - 38
  - 47
  - 231
  - 185
  - 230
  - 98
  - 17
  - 45
  - 187
  - 128
  - 124
  - 253
  - 156
  sequence_number: 46
request: []
response: []

```
                
    
            
### Step 3 - Configure client to send introspection metadata
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp introspect_services_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 param set /introspection_client client_configure_introspection metadata
```
                
```bash
# StdIn - terminal 2
ros2 topic echo /add_two_ints/_service_event
```
                
    

    
        
#### Expected Results
        
        
```bash
# StdOut - terminal 1
[introspection_service-1] a: 2 b: 3
[introspection_client-2] [INFO] [1680585324.964598437] [introspection_client]: Result of add_two_ints: 5

```
                
```bash
# StdOut - terminal 2
info:
  event_type: 0
  stamp:
    sec: 1680586637
    nanosec: 158736272
  client_gid:
  - 1
  - 16
  - 34
  - 142
  - 114
  - 128
  - 207
  - 116
  - 98
  - 246
  - 204
  - 52
  - 0
  - 0
  - 0
  - 7
  sequence_number: 48
request: []
response: []
---
info:
  event_type: 3
  stamp:
    sec: 1680586637
    nanosec: 159157853
  client_gid:
  - 1
  - 16
  - 34
  - 142
  - 114
  - 128
  - 207
  - 116
  - 98
  - 246
  - 204
  - 52
  - 0
  - 0
  - 0
  - 7
  sequence_number: 48
request: []
response: []

```

ganatrask, 2024-05-01:
I tested this, everything work as expected. 

Can you close this?

ganatrask, 2024-05-01:
Here are some screenshots of RQT working smoothly

[image]
[image]
[image]
[image]

closed 2024-05-02 by wjwwood