osrf/ros2_test_cases #1322: Dummy Map Server executables

marcoag, 2024-04-30:
## Configuration

* OS: Windows
* Chip set: AMD64
* Build type: Binary
* DDS: FastDDS

## Links

* [Dummy Map Server](https://github.com/ros2/demos/blob/jazzy/dummy_robot/dummy_map_server/README.md#run)

## Process

    
            
### The executables in the README are present and work as expected.

Trushant-Adeshara-UM, 2024-05-15:
I would like to take up this ticket, can you please assign it to me?

Trushant-Adeshara-UM, 2024-05-15:
### Environment:
OS: Windows 11

### Result:
The node ```dummy_map_server``` was successfully generated and it publishes on the topic ```/map```, as shown below:

[image]

ahcorde, 2024-05-15:
Thank you for testing

closed 2024-05-15 by ahcorde