osrf/ros2_test_cases #1324: Dummy Sensors executables

marcoag, 2024-04-30:
## Configuration

* OS: Windows
* Chip set: AMD64
* Build type: Binary
* DDS: FastDDS

## Links

* [Dummy Sensors](https://github.com/ros2/demos/blob/jazzy/dummy_robot/dummy_sensors/README.md#run)

## Process

    
            
### The executables in the README are present and work as expected.

Trushant-Adeshara-UM, 2024-05-15:
I would like to take up this ticket, can you please assign it to me?

Trushant-Adeshara-UM, 2024-05-15:
### Environment:
OS: Windows 11

### Result 1: Dummy Laser
The node ```dummy_laser``` successfully generated and it published message of the type ```sensor_msgs/msg/LaserScan``` on the topic ```/scan```, as shown below:

[image]

### Result 2: Dummy Joint States
The node ```dummy_joint_states``` was successfully generated and it published message of the type ```sensor_msgs/msg/JointState``` on the topic ```/joint_states```, as shown below:

[image]

ahcorde, 2024-05-15:
Thank you for testing

closed 2024-05-15 by ahcorde