osrf/ros2_test_cases #1326: Image Tools executables

marcoag, 2024-04-30:
## Configuration

* OS: Windows
* Chip set: AMD64
* Build type: Binary
* DDS: FastDDS

## Links

* [Image Tools](https://github.com/ros2/demos/blob/jazzy/image_tools/README.md#run)

## Process

    
            
### The executables in the README are present and work as expected.

Trushant-Adeshara-UM, 2024-05-15:
I would like to take up this ticket, can you please assign it to me?

Trushant-Adeshara-UM, 2024-05-15:
### Environment:
OS: Windows 11

### Result: Capture and Show Image
The node ```cam2image``` was successfully generated, and it published message of the type ```sensor_msg/msg/Image``` on the topic ```/image``` which was then subscribed by node ```showimage``` to visualize, as shown below:

[image]

ahcorde, 2024-05-15:
Thank you for testing

closed 2024-05-15 by ahcorde