osrf/ros2_test_cases #1328: Intra-process executables

marcoag, 2024-04-30:
## Configuration

* OS: Windows
* Chip set: AMD64
* Build type: Binary
* DDS: FastDDS

## Links

* [Intra-process demos](https://github.com/ros2/demos/blob/jazzy/intra_process_demo/README.md)

## Process

    
            
### The executables in the README are present and work as expected.

Trushant-Adeshara-UM, 2024-05-15:
I would like to take up this ticket, can you please assign it to me?

Trushant-Adeshara-UM, 2024-05-15:
### Environment:
OS: Windows 11

### Result 1: Two Node Pipeline 
The executable ```two_node_pipeline``` is present and functions as expected, as demonstrated in the snapshot below:

[image]

### Result 2: Cyclic Pipeline 
The executable ```cyclic_pipeline``` is present and functions as expected, as demonstrated in the snapshot below:

[image]

### Result 3: Image Pipeline All In One 
The executable ```image_pipeline_all_in_one``` is present and functions as expected, as demonstrated in the snapshot below:

[image]

### Result 4: Image Pipeline All Separately 
Note: The second executable is ```watermark_image``` and not ```watermarked_image```. A PR is generated which can be found [here.](https://github.com/ros2/demos/pull/689)

Other two executables, ```camera_node``` and ```image_view_node``` are present and their working is demonstrated in the snapshot below:

[image]

### Result 5: Image Pipeline With Two Image Views 
The executable ```image_pipeline_with_two_image_view``` is present and functions as expected, as demonstrated in the snapshot below:

[image]

ahcorde, 2024-05-15:
Thank you for testing

closed 2024-05-15 by ahcorde