osrf/ros2_test_cases #1330: Launch testing

marcoag, 2024-04-30:
## Configuration

* OS: Windows
* Chip set: AMD64
* Build type: Binary
* DDS: FastDDS

## Links

* [Launch testing examples](https://github.com/ros2/launch/blob/jazzy/launch_testing/README.md#examples)

## Process

    
            
### Step 1 - Clone the example repository
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
mkdir -p ~/launch_ws/src
```
                
```bash
# StdIn - terminal 1
git clone https://github.com/ros2/launch -b jazzy ~/launch_ws/src/launch
```
                
    

    
    
            
### Step 2 - Test `hello_world_launch_test.py`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/hello_world_launch_test.py
```
                
    

    
    
            
### Step 3 - Test `good_proc_launch_test.py`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/good_proc_launch_test.py
```
                
    

    
    
            
### Step 4 - Test `terminating_proc_launch_test.py`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/terminating_proc_launch_test.py
```
                
    

    
    
            
### Step 5 - Test `args_launch_test.py`
            
        

    
    

    
        
        
Show the arguments
                
```bash
# StdIn - terminal 1
launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/args_launch_test.py --show-args
```
                
Run the test
                
```bash
# StdIn - terminal 1
launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/args_launch_test.py dut_arg:=hey
```
                
    

    
    
            
### Step 6 - Test `context_launch_test.py`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/context_launch_test.py
```

pradyum, 2024-05-05:
## Setup

#### Installation Changes

1. openssl v1.1.1 - downloaded through choco

```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

The windows binary installation does not have steps to install git and pytest which which are required to run the launch testing. This test passed after install git and pytest.

ahcorde, 2024-05-06:
@pradyum do you mind to add then to the install instructions ?

pradyum, 2024-05-07:
@ahcorde Yeah, I have added **pytest** python dependency to **windows-install-prerequisites**

clalancette, 2024-05-16:
This was added to the documentation, so closing this out.  Thanks for testing!

closed 2024-05-16 by clalancette