osrf/ros2_test_cases #1335: Lifecycle demo

marcoag, 2024-04-30:
## Configuration

* Chip set: AMD64
* DDS: CycloneDDS
* Build type: Binary
* OS: Windows

## Links

* [Reference URL](https://github.com/ros2/demos/blob/jazzy/lifecycle/README.rst)

## Process

    
            
### Run files individually
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run lifecycle lifecycle_talker
```
                
[image]
                
```bash
# StdIn - terminal 2
ros2 run lifecycle lifecycle_listener
```
                
[image]
                
```bash
# StdIn - terminal 3
ros2 run lifecycle lifecycle_service_client
```
                
[image]
                
    

    
        
#### Expected Result
        
        
The images match.

pradyum, 2024-05-05:
## Setup

#### Installation Changes

1. openssl v1.1.1 - downloaded through choco

```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

#### Error
```
C:\>ros2 run lifecycle lifecycle_service_client
[ERROR] [1714927506.402826800] [lc_client]: Server time out while getting current state for node lc_talker
```

clalancette, 2024-05-07:
> C:\>ros2 run lifecycle lifecycle_service_client
> [ERROR] [1714927506.402826800] [lc_client]: Server time out while getting current state for node lc_talker

Oh, hm.  This seems to work for me on Windows.  Did you have both `lifecycle_talker` and `lifecycle_listener` running prior to running `lifecycle_service_client`?  All 3 need to be running for this test to be successful.

pradyum, 2024-05-08:
Sorry, The lifecycle transition are triggered 1 -> 3 -> 4 -> 3 -> 4 -> 2 -> 5 are working. The messages are communicated between talker and listener. This test is working and passed.

closed 2024-05-08 by ahcorde

clalancette, 2024-05-08:
> Sorry, The lifecycle transition are triggered 1 -> 3 -> 4 -> 3 -> 4 -> 2 -> 5 are working. The messages are communicated between talker and listener. This test is working and passed.

Thanks for double checking!