osrf/ros2_test_cases #1391: Executables in `rosbag2`

marcoag, 2024-04-30:
## Configuration

* Chip set: AMD64
* DDS: CycloneDDS
* Build type: Binary
* OS: Windows

## Process

    
            
### Step 1 - Check `ros2 bag record -a`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 bag record -a
```
                
    

    
    
            
### Step 2 - Check `ros2 bag record <topic1> <topic2> <topic3>`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 bag record <topic1> <topic2> <topic3>
```
                
    

    
    
            
### Step 3 - Check `ros2 bag info <bag_file>`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 bag info <bag_file>
```
                
    

    
    
            
### Step 4 - Check `ros2 bag play <bag_file>`
            
        

    
    

    
        
        
```bash
# StdIn - terminal 1
ros2 bag play <bag_file>
```

pradyum, 2024-05-05:
## Setup

#### Installation Changes

1. openssl v1.1.1 - downloaded through choco

```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

#### Error

There is timestamp error.

```
C:\Users\pc\Desktop>ros2 bag info rosbag2_2024_05_05-09_14_21

Files:             rosbag2_2024_05_05-09_14_21_0.mcap
Bag size:          25.9 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          0.0s
Start:             Dec 31 1969 16:00:00.0 (0.0)
End:               Dec 31 1969 16:00:00.0 (0.0)
Messages:          21
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 4 | Serialization Format: cdr
                   Topic: /events/write_split | Type: rosbag2_interfaces/msg/WriteSplitEvent | Count: 0 | Serialization Format: cdr
                   Topic: /parameter_events | Type: rcl_interfaces/msg/ParameterEvent | Count: 0 | Serialization Format: cdr
                   Topic: /rosout | Type: rcl_interfaces/msg/Log | Count: 17 | Serialization Format: cdr
Service:           0
Service information:
```

In my previous testing with powershell this worked #1390, But not in command prompt which I tested with now.

marcoag, 2024-06-26:
This was reported and solved through https://github.com/ros2/rosbag2/issues/1638. Thanks for the help here @pradyum.

closed 2024-06-26 by marcoag