osrf/ros2_test_cases #1392: Check action Cli

marcoag, 2024-04-30:
## Configuration

* OS: Windows
* Chip set: AMD64
* Build type: Binary
* DDS: FastDDS

## Process

    
            
### Step 1 - Check info
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
```
                
```bash
# StdIn - terminal 2
ros2 action info /fibonacci
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
Action: /fibonacci
Action clients: 0
Action servers: 1
  /minimal_action_server

```
                
    
            
### Step 2 - Check list
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
```
                
```bash
# StdIn - terminal 2
ros2 action list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/fibonacci
```
                
    
            
### Step 3 - Check send_goal
            
        

    
    

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
```
                
```bash
# StdIn - terminal 2
ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci order:\ 2
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
Waiting for an action server to become available...
Sending goal:
    order: 2

Goal accepted with ID: 0110dbd59e804374bff8e4900c3201e8

Result:
    sequence:
- 0
- 1
- 1

Goal finished with status: SUCCEEDED

```

pradyum, 2024-05-05:
## Setup

#### Installation Changes

1. openssl v1.1.1 - downloaded through choco

```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

In Step 3 I had to send_goal as
```
ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci "order: 2"
```

And then got the expected result. This test has passed.

marcoag, 2024-05-06:
Sounds good, thanks!

closed 2024-05-06 by marcoag