osrf/ros2_test_cases #1395: Check ROS 2 bag CLI

marcoag, 2024-04-30:
## Configuration

* Chip set: AMD64
* DDS: CycloneDDS
* Build type: Binary
* OS: Windows

## Process

    
            
### Step 1 - Check record
            
        

    
    

    
        
            
#### Actions
            
        
        
You should stop the talker once the recording is stopped.
                
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp talker
```
                
Run for a few seconds and then hit Ctrl-C to stop the bag recording.
                
```bash
# StdIn - terminal 2
ros2 bag record /chatter -o my-bag
```
                
    

    
        
#### Expected Result
        
        
You should see a folder named my-bag in the current directory
                
    
            
### Step 2 - Check info
            
        

    
    

    
        
            
#### Actions
            
        
        
Continuing from the previous check.
                
```bash
# StdIn - terminal 1
ros2 bag info my-bag
```
                
    

    
        
#### Expected Results
        
        
Something like the following
                
```bash
# StdOut - terminal 1
Files:             my-bag_0.mcap
Bag size:          3.0 KiB
Storage id:        mcap
Duration:          3.999s
Start:             Apr  2 2023 13:45:47.834 (1680414347.834)
End:               Apr  2 2023 13:45:51.834 (1680414351.834)
Messages:          5
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 5 | Serialization Format: cdr

```
                
    
            
### Step 3 - Check play
            
        

    
    

    
        
            
#### Actions
            
        
        
Continuing from the previous two checks. Make sure that the talker is stopped.
                
The `-l` option loops the bag. You should kill this process eventually with Ctrl-C.
                
```bash
# StdIn - terminal 1
ros2 bag play -l my-bag
```
                
```bash
# StdIn - terminal 2
ros2 topic echo /chatter
```
                
    

    
        
#### Expected Results
        
        
Something like the following.
                
```bash
# StdOut - terminal 2
---
data: 'Hello World: 77'
---
data: 'Hello World: 78'
---
data: 'Hello World: 75'
---
data: 'Hello World: 76'
---
data: 'Hello World: 77'
---

```

pradyum, 2024-05-05:
## Setup

#### Installation Changes

1. openssl v1.1.1 - downloaded through choco

```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

#### Error

The previous test which I did in powershell has passed. But, this test which is done in command prompt has failed. There is 0.0s duration and same start and end timestamp which is predated to initial timestamp = 0. 

```
C:\Users\pc\Desktop>ros2 bag info my-bag

Files:             my-bag_0.mcap
Bag size:          5.7 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          0.0s
Start:             Dec 31 1969 16:00:00.0 (0.0)
End:               Dec 31 1969 16:00:00.0 (0.0)
Messages:          12
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 12 | Serialization Format: cdr
Service:           0
Service information:
```

clalancette, 2024-05-06:
This is another instance of https://github.com/ros2/rosbag2/issues/1638

closed 2024-06-26 by marcoag