osrf/ros2_test_cases #1401: Check ROS 2 doctor CLI

marcoag, 2024-04-30:
## Configuration

* Chip set: AMD64
* DDS: CycloneDDS
* Build type: Binary
* OS: Windows

## Process

    
            
### Step 1 - Check `-r`
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 doctor -r
```
                
    

    
        
#### Expected Result
        
        
A large report is generated with several sections.
                
    
            
### Step 2 - Check `-rf`
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 doctor -rf
```
                
    

    
        
#### Expected Result
        
        
A lot of checks and a summary of the checks is given (e.g., 'All 5 checks passed').
                
    
            
### Step 3 - Check `-iw`
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 doctor -iw
```
                
    

    
        
#### Expected Result
        
        
A lot of checks and a summary of the checks is given (e.g., '2/5 checks passed').
                
    
            
### Step 4 - Check hello
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 doctor hello
```
                
    

    
        
#### Expected Results
        
        
You'll see repeated messages that look somewhat like the following.
                
```bash
# StdOut - terminal 1
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 11
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------

```

pradyum, 2024-05-05:
## Setup

#### Installation Changes

1. openssl v1.1.1 - downloaded through choco

```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

The checks are working as expected and this test has passed.

marcoag, 2024-05-06:
Great, thanks!

closed 2024-05-06 by marcoag