osrf/ros2_test_cases #1406: Check ROS 2 launch CLI

marcoag, 2024-04-30:
## Configuration

* OS: Windows
* Chip set: AMD64
* Build type: Binary
* DDS: FastDDS

## Process

    
            
### Step 1 - Check launch package
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener_launch.py
```
                
    

    
        
#### Expected Results
        
        
You should see something like the following
                
```bash
# StdOut - terminal 1
[INFO] [launch]: All log files can be found below /home/parallels/.ros/log/2022-03-21-17-38-54-737866-parallels-Parallels-Virtual-Platform-3717
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [3722]
[INFO] [listener-2]: process started with pid [3724]
[talker-1] [INFO] [1647909535.937731548] [talker]: Publishing: 'Hello World: 1'
[listener-2] [INFO] [1647909535.938337387] [listener]: I heard: [Hello World: 1]
[talker-1] [INFO] [1647909536.937932068] [talker]: Publishing: 'Hello World: 2'
[listener-2] [INFO] [1647909536.938125486] [listener]: I heard: [Hello World: 2]
[talker-1] [INFO] [1647909537.936162331] [talker]: Publishing: 'Hello World: 3'
[listener-2] [INFO] [1647909537.936366391] [listener]: I heard: [Hello World: 3]
...

```
                
    
            
### Step 2 - Check launch launch file directly
            
        

    
    

    
        
            
#### Actions
            
        
        
Download a launch file, for example with `curl`.
                
```bash
# StdIn - terminal 1
curl https://raw.githubusercontent.com/ros2/demos/master/demo_nodes_cpp/launch/topics/talker_listener_launch.py > /tmp/talker_listener_launch.py
```
                
```bash
# StdIn - terminal 1
ros2 launch /tmp/talker_listener_launch.py
```
                
    

    
        
#### Expected Results
        
        
You should see something like the following
                
```bash
# StdOut - terminal 1
[INFO] [launch]: All log files can be found below /home/parallels/.ros/log/2022-03-21-17-38-54-737866-parallels-Parallels-Virtual-Platform-3717
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [3722]
[INFO] [listener-2]: process started with pid [3724]
[talker-1] [INFO] [1647909535.937731548] [talker]: Publishing: 'Hello World: 1'
[listener-2] [INFO] [1647909535.938337387] [listener]: I heard: [Hello World: 1]
[talker-1] [INFO] [1647909536.937932068] [talker]: Publishing: 'Hello World: 2'
[listener-2] [INFO] [1647909536.938125486] [listener]: I heard: [Hello World: 2]
[talker-1] [INFO] [1647909537.936162331] [talker]: Publishing: 'Hello World: 3'
[listener-2] [INFO] [1647909537.936366391] [listener]: I heard: [Hello World: 3]
...

```

pradyum, 2024-05-05:
## Setup

#### Installation Changes

1. openssl v1.1.1 - downloaded through choco

```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

#### Error

The curl command in Step 2 output the status of http request:
```
StatusCode        : 200
StatusDescription : OK
Content           : # Copyright 2018 Open Source Robotics Foundation, Inc.
                    #
                    # Licensed under the Apache License, Version 2.0 (the "License");
                    # you may not use this file except in compliance with the License.
                    # You may ...
RawContent        : HTTP/1.1 200 OK
                    Connection: keep-alive
                    Content-Security-Policy: default-src 'none'; style-src
                    'unsafe-inline'; sandbox
                    Strict-Transport-Security: max-age=31536000
                    X-Content-Type-Options: nosniff
                    ...
Forms             : {}
Headers           : {[Connection, keep-alive], [Content-Security-Policy, default-src
                    'none'; style-src 'unsafe-inline'; sandbox],
                    [Strict-Transport-Security, max-age=31536000],
                    [X-Content-Type-Options, nosniff]...}
Images            : {}
InputFields       : {}
Links             : {}
ParsedHtml        : System.__ComObject
RawContentLength  : 1001
```

 After downloading the file manually and executing it I got the expected results. The issue is with curl and not with ROS 2 launch CLI. So, this test is passed.

marcoag, 2024-05-06:
Sounds good, thanks!

closed 2024-05-06 by marcoag