osrf/ros2_test_cases #1415: Check ROS 2 Pkg CLI

marcoag, 2024-04-30:
## Configuration

* Chip set: AMD64
* DDS: CycloneDDS
* Build type: Binary
* OS: Windows

## Process

    
            
### Step 1 - Check create
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 pkg create my-pkg
```
                
    

    
        
#### Expected Result
        
        
Creates a folder named `my-pkg`, which has a `package.xml` file, `CMakeLists.txt` file, `src` folder, and `include` folder.
                
    
            
### Step 2 - Check executables
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 pkg executables turtlesim
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 1
turtlesim draw_square
turtlesim mimic
turtlesim turtle_teleop_key
turtlesim turtlesim_node

```
                
    
            
### Step 3 - Check list
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 pkg list
```
                
    

    
        
#### Expected Results
        
        
Lists all the packages, for example.
                
```bash
# StdOut - terminal 1
action_msgs
action_tutorials_cpp
action_tutorials_interfaces
...
visualization_msgs
yaml_cpp_vendor
zstd_vendor

```
                
    
            
### Step 4 - Check prefix
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 pkg prefix turtlesim
```
                
    

    
        
#### Expected Result
        
        
This varies depending on your install type. For example, from binaries on Ubuntu, it will be `/opt/ros/jazzy`
                
    
            
### Step 5 - Check xml
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 pkg xml turtlesim
```
                
    

    
        
#### Expected Result
        
        
The full `package.xml` of the `turtlesim` package is printed to stdout.

pradyum, 2024-05-05:
## Setup

#### Installation Changes

1. openssl v1.1.1 - downloaded through choco

```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

A package change has to be made to make this working as said in #1414. After the changes the checks are working as expected and the test passes.

marcoag, 2024-05-06:
Great, thanks!

closed 2024-05-06 by marcoag