osrf/ros2_test_cases #1416: Check ROS 2 run CLI

marcoag, 2024-04-30:
## Configuration

* OS: Windows
* Chip set: AMD64
* Build type: Binary
* DDS: FastDDS

## Process

    
            
### Step 1 - Check run
            
        

    
    

    
        
            
#### Actions
            
        
        
You'll have to kill this command with Ctrl-C.
                
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp talker
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 1
[INFO] [1649123362.906256690] [talker]: Publishing: 'Hello World: 1'
[INFO] [1649123363.906469087] [talker]: Publishing: 'Hello World: 2'
[INFO] [1649123364.905960807] [talker]: Publishing: 'Hello World: 3'
...

```
                
    
            
### Step 2 - Check run with prefix
            
        

    
    

    
        
            
#### Action
            
        
        
```bash
# StdIn - terminal 1
ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker
```
                
    

    
        
#### Expected Result
        
        
This runs the talker node in gdb. Once the talker is running, you can hit Ctrl-C to get to the console and then type `q` to quit gdb.

Trushant-Adeshara-UM, 2024-05-14:
I would like to take up this ticket, can you please assign it to me?

Trushant-Adeshara-UM, 2024-05-14:
For the issue mentioned, Step 1 checks out. However, regarding Step 2, it appears necessary to install the debug version of the binaries. It would be beneficial for the documentation to specify this requirement to ensure clarity and ease of setup for future users.

Trushant-Adeshara-UM, 2024-05-15:
I am experiencing issues with **ros.exe** failing when using the debug codebase. Has a solution already been found for this problem?

marcoag, 2024-05-15:
@Trushant-Adeshara-UM can you please be more specific on what you mean by "issues"?

closed 2024-06-26 by marcoag