osrf/ros2_test_cases #1418: Check ROS 2 security CLI on Windows

marcoag, 2024-04-30:
## Configuration

* OS: Windows
* Chip set: AMD64
* Build type: Binary
* DDS: FastDDS

## Links

* [Tutorial URL](https://github.com/ros2/sros2/blob/master/SROS2_Windows.md)

## Process

    
            
### The Linux tutorials work

mikaelarguedas, 2024-05-06:
In case it's useful to the person testing this:
On Linux there was an issue in the Access control section as the svn command didnt work.
The workaround is to clone the entire sros2 repository in a temporary location: https://github.com/ros2/sros2/pull/295

If you face the same issue and a similar workaround works for you, could you please consider opening a PR modifying https://github.com/ros2/sros2/blob/ca6bb12cc650b73e7ccfc0fa789d8b49358d44ad/SROS2_Windows.md?plain=1#L24-L35 and https://github.com/ros2/sros2/blob/ca6bb12cc650b73e7ccfc0fa789d8b49358d44ad/SROS2_Windows.md?plain=1#L169-L178 accordingly ?

Thanks :pray:

closed 2024-06-26 by marcoag

mikaelarguedas, 2024-06-26:
@marcoag was this tested? Is it possible to open the corresponding readme PR based on the result of the testing?

marcoag, 2024-06-26:
This issue was not picked up and tested, if that's what you mean. Regarding your comment I'll add it to our list of issues to review and check for the readme PR.