osrf/ros2_test_cases #1427: ros2cli on a local machine

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family

## Configuration

* Chip set: AMD64
* DDS: CycloneDDS
* OS: Ubuntu Noble 24.04
* Build type: Debian

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Help text
            
        

    
Long-form help text should be available when invoking `ros2 --help` as well as `ros2 <command> --help`

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 -h
```
                
```bash
# StdIn - terminal 1
ros2 --help
```
                
```bash
# StdIn - terminal 1
ros2 topic -h
```
                
```bash
# StdIn - terminal 1
ros2 topic --help
```
                
    

    
        
#### Expected Result
        
        
Long-form help is available.
                
    
            
### Step 2 - Tab completion
            
        

    
When using a supported shell (i.e. bash), tab completion should work for both the bare executable to list commands and for listing sub-commands, like `ros2 <tab>` or `ros2 <command> tab`

        

    
        
            
#### Actions
            
        
        
Lists the commands
                
```bash
# StdIn - terminal 1
ros2 <tab><tab>
```
                
Completes the word `launch`
                
```bash
# StdIn - terminal 1
ros2 lau<tab>
```
                
Completes until `demo_nodes_`
                
```bash
# StdIn - terminal 1
ros2 launch dem<tab>
```
                
Completes the package `demo_nodes_cpp`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_c<tab>
```
                
Lists launch files in `demo_nodes_cpp`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp <tab><tab>
```
                
Suggests the launch files name `talker_listener`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp ta<tab>
```
                
Lists several launch files that start with `talker_listener`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener<tab><tab>
```
                
Completes the launch files name `talker_listener_launch.py`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener_launch.p<tab>
```
                
    

    
        
#### Expected Results
        
        
You eventually get to the full command
                
Stdin: "ros2 launch demo_nodes_cpp talker_listener_launch.py"
                
    
            
### Step 3 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 node list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/listener
/talker

```
                
    
            
### Step 4 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/chatter
/parameter_events
/rosout

```
                
    
            
### Step 5 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp add_two_ints_server
```
                
```bash
# StdIn - terminal 2
ros2 service list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically

```
                
    
            
### Step 6 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run turtlesim turtlesim_node
```
                
```bash
# StdIn - terminal 2
ros2 param list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/turtlesim:
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time

```

fujitatomoya, 2024-05-02:
@marcoag sign me up for this.

fujitatomoya, 2024-05-02:
### test result console log

<details>
  <summary>Step 2 - Tab completion</summary>

```bash
root@51cdd59e1f3e:~# ros2 
action                          extension_points                multicast                       security
bag                             extensions                      node                            service
component                       interface                       param                           topic
daemon                          launch                          pkg                             --use-python-default-buffering
doctor                          lifecycle                       run                             wtf
root@51cdd59e1f3e:~# ros2 launch 
usage: ros2 launch [-h] [-n] [-d] [-p | -s] [-a] [--launch-prefix LAUNCH_PREFIX] [--launch-prefix-filter LAUNCH_PREFIX_FILTER]
                   package_name [launch_file_name] [launch_arguments ...]
ros2 launch: error: the following arguments are required: package_name, launch_arguments
root@51cdd59e1f3e:~# ros2 launch demo_nodes_
Package 'demo_nodes_' not found: "package 'demo_nodes_' not found, searching: ['/opt/ros/jazzy']"
root@51cdd59e1f3e:~# ros2 launch demo_nodes_cpp 
-a                                       --launch-prefix-filter                   --show-args
add_two_ints_async_launch.py             -n                                       --show-arguments
add_two_ints_async_launch.xml            --noninteractive                         talker_listener_best_effort_launch.py
add_two_ints_launch.py                   -p                                       talker_listener_best_effort_launch.xml
add_two_ints_launch.xml                  package.xml                              talker_listener_best_effort_launch.yaml
-d                                       --print                                  talker_listener_launch.py
--debug                                  --print-description                      talker_listener_launch.xml
introspect_services_launch.py            -s                                       talker_listener_launch.yaml
--launch-prefix                          --show-all-subprocesses-output           
root@51cdd59e1f3e:~# ros2 launch demo_nodes_cpp 
file 'None' was not found in the share directory of package 'demo_nodes_cpp' which is at '/opt/ros/jazzy/share/demo_nodes_cpp'
root@51cdd59e1f3e:~# ros2 launch demo_nodes_cpp talker_listener_
file 'talker_listener_' was not found in the share directory of package 'demo_nodes_cpp' which is at '/opt/ros/jazzy/share/demo_nodes_cpp'
root@51cdd59e1f3e:~# ros2 launch demo_nodes_cpp talker_listener_
talker_listener_best_effort_launch.py    talker_listener_best_effort_launch.yaml  talker_listener_launch.xml
talker_listener_best_effort_launch.xml   talker_listener_launch.py                talker_listener_launch.yaml
root@51cdd59e1f3e:~# ros2 launch demo_nodes_cpp talker_listener_
talker_listener_best_effort_launch.py    talker_listener_best_effort_launch.yaml  talker_listener_launch.xml
talker_listener_best_effort_launch.xml   talker_listener_launch.py                talker_listener_launch.yaml
root@51cdd59e1f3e:~# ros2 launch demo_nodes_cpp talker_listener_launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-05-01-17-07-22-653420-51cdd59e1f3e-19426
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [19429]
[INFO] [listener-2]: process started with pid [19430]
[talker-1] [INFO] [1714608443.710569109] [talker]: Publishing: 'Hello World: 1'
[listener-2] [INFO] [1714608443.711149371] [listener]: I heard: [Hello World: 1]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[listener-2] [INFO] [1714608443.896012569] [rclcpp]: signal_handler(signum=2)
[talker-1] [INFO] [1714608443.896020698] [rclcpp]: signal_handler(signum=2)
[INFO] [listener-2]: process has finished cleanly [pid 19430]
[INFO] [talker-1]: process has finished cleanly [pid 19429]
```

</details>

<details>
  <summary>Step 3 - Node discovery</summary>

```bash
root@51cdd59e1f3e:~# ros2 node list
/listener
/talker
```

</details>

<details>
  <summary>Step 4 - Topic discovery</summary>

```bash
root@51cdd59e1f3e:~# ros2 topic list
/chatter
/parameter_events
/rosout
```

</details>

<details>
  <summary>Step 5 - Service discovery</summary>

```bash
root@51cdd59e1f3e:~# ros2 service list
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/get_type_description
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically
```

</details>

<details>
  <summary>Step 6 - Parameter discovery</summary>

```bash
root@51cdd59e1f3e:~# ros2 param list
/turtlesim:
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
```

</details>

fujitatomoya, 2024-05-02:
@marcoag i think we can close this issue.

closed 2024-05-02 by wjwwood