osrf/ros2_test_cases #1431: ros2cli on a local machine

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family

## Configuration

* DDS: ConnextDDS
* Chip set: ARM64
* Build type: Debian
* OS: Ubuntu Noble 24.04

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Help text
            
        

    
Long-form help text should be available when invoking `ros2 --help` as well as `ros2 <command> --help`

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 -h
```
                
```bash
# StdIn - terminal 1
ros2 --help
```
                
```bash
# StdIn - terminal 1
ros2 topic -h
```
                
```bash
# StdIn - terminal 1
ros2 topic --help
```
                
    

    
        
#### Expected Result
        
        
Long-form help is available.
                
    
            
### Step 2 - Tab completion
            
        

    
When using a supported shell (i.e. bash), tab completion should work for both the bare executable to list commands and for listing sub-commands, like `ros2 <tab>` or `ros2 <command> tab`

        

    
        
            
#### Actions
            
        
        
Lists the commands
                
```bash
# StdIn - terminal 1
ros2 <tab><tab>
```
                
Completes the word `launch`
                
```bash
# StdIn - terminal 1
ros2 lau<tab>
```
                
Completes until `demo_nodes_`
                
```bash
# StdIn - terminal 1
ros2 launch dem<tab>
```
                
Completes the package `demo_nodes_cpp`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_c<tab>
```
                
Lists launch files in `demo_nodes_cpp`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp <tab><tab>
```
                
Suggests the launch files name `talker_listener`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp ta<tab>
```
                
Lists several launch files that start with `talker_listener`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener<tab><tab>
```
                
Completes the launch files name `talker_listener_launch.py`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener_launch.p<tab>
```
                
    

    
        
#### Expected Results
        
        
You eventually get to the full command
                
Stdin: "ros2 launch demo_nodes_cpp talker_listener_launch.py"
                
    
            
### Step 3 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 node list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/listener
/talker

```
                
    
            
### Step 4 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/chatter
/parameter_events
/rosout

```
                
    
            
### Step 5 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp add_two_ints_server
```
                
```bash
# StdIn - terminal 2
ros2 service list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically

```
                
    
            
### Step 6 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run turtlesim turtlesim_node
```
                
```bash
# StdIn - terminal 2
ros2 param list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/turtlesim:
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time

```

pradyum, 2024-05-03:
## Test Environment

```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.8.0-31-generic-aarch64-with-glibc2.39
release          : 6.8.0-31-generic
processor        : aarch64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```

## Result

I am getting the expected results as per each actions. This test is passed.

wep21, 2024-05-03:
`rmw_connextdds` installed by debian doesn't have any library target.
```
root@b86ef71be53f:/# dpkg -L ros-jazzy-rmw-connextdds
/.
/opt
/opt/ros
/opt/ros/jazzy
/opt/ros/jazzy/share
/opt/ros/jazzy/share/ament_index
/opt/ros/jazzy/share/ament_index/resource_index
/opt/ros/jazzy/share/ament_index/resource_index/package_run_dependencies
/opt/ros/jazzy/share/ament_index/resource_index/package_run_dependencies/rmw_connextdds
/opt/ros/jazzy/share/ament_index/resource_index/packages
/opt/ros/jazzy/share/ament_index/resource_index/packages/rmw_connextdds
/opt/ros/jazzy/share/ament_index/resource_index/parent_prefix_path
/opt/ros/jazzy/share/ament_index/resource_index/parent_prefix_path/rmw_connextdds
/opt/ros/jazzy/share/rmw_connextdds
/opt/ros/jazzy/share/rmw_connextdds/cmake
/opt/ros/jazzy/share/rmw_connextdds/cmake/rmw_connextddsConfig-version.cmake
/opt/ros/jazzy/share/rmw_connextdds/cmake/rmw_connextddsConfig.cmake
/opt/ros/jazzy/share/rmw_connextdds/environment
/opt/ros/jazzy/share/rmw_connextdds/environment/ament_prefix_path.dsv
/opt/ros/jazzy/share/rmw_connextdds/environment/ament_prefix_path.sh
/opt/ros/jazzy/share/rmw_connextdds/environment/path.dsv
/opt/ros/jazzy/share/rmw_connextdds/environment/path.sh
/opt/ros/jazzy/share/rmw_connextdds/local_setup.bash
/opt/ros/jazzy/share/rmw_connextdds/local_setup.dsv
/opt/ros/jazzy/share/rmw_connextdds/local_setup.sh
/opt/ros/jazzy/share/rmw_connextdds/local_setup.zsh
/opt/ros/jazzy/share/rmw_connextdds/package.dsv
/opt/ros/jazzy/share/rmw_connextdds/package.xml
/usr
/usr/share
/usr/share/doc
/usr/share/doc/ros-jazzy-rmw-connextdds
/usr/share/doc/ros-jazzy-rmw-connextdds/changelog.Debian.gz
/usr/share/doc/ros-jazzy-rmw-connextdds/copyright
```
So, all ros2 command fails like
```
root@b86ef71be53f:/# ros2
[ERROR] [1714732515.120176148] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to load shared library 'librmw_connextdds.so' due to dlopen error: librmw_connextdds.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99, at ./src/functions.cpp:65', exiting with 1., at ./src/rcl/rmw_implementation_identifier_check.c:143
```

clalancette, 2024-05-03:
> `rmw_connextdds` installed by debian doesn't have any library target.

Yeah, that is correct; on ARM64 we don't have ConnextDDS support.  That was a bug in our scripts to generate the tests.

I'm going to close this out, thanks for testing.

closed 2024-05-03 by clalancette