osrf/ros2_test_cases #1436: ros2cli on a local machine

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family

## Configuration

* OS: Ubuntu Noble 24.04
* Build type: Binary
* Chip set: ARM64
* DDS: CycloneDDS

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Help text
            
        

    
Long-form help text should be available when invoking `ros2 --help` as well as `ros2 <command> --help`

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 -h
```
                
```bash
# StdIn - terminal 1
ros2 --help
```
                
```bash
# StdIn - terminal 1
ros2 topic -h
```
                
```bash
# StdIn - terminal 1
ros2 topic --help
```
                
    

    
        
#### Expected Result
        
        
Long-form help is available.
                
    
            
### Step 2 - Tab completion
            
        

    
When using a supported shell (i.e. bash), tab completion should work for both the bare executable to list commands and for listing sub-commands, like `ros2 <tab>` or `ros2 <command> tab`

        

    
        
            
#### Actions
            
        
        
Lists the commands
                
```bash
# StdIn - terminal 1
ros2 <tab><tab>
```
                
Completes the word `launch`
                
```bash
# StdIn - terminal 1
ros2 lau<tab>
```
                
Completes until `demo_nodes_`
                
```bash
# StdIn - terminal 1
ros2 launch dem<tab>
```
                
Completes the package `demo_nodes_cpp`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_c<tab>
```
                
Lists launch files in `demo_nodes_cpp`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp <tab><tab>
```
                
Suggests the launch files name `talker_listener`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp ta<tab>
```
                
Lists several launch files that start with `talker_listener`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener<tab><tab>
```
                
Completes the launch files name `talker_listener_launch.py`
                
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener_launch.p<tab>
```
                
    

    
        
#### Expected Results
        
        
You eventually get to the full command
                
Stdin: "ros2 launch demo_nodes_cpp talker_listener_launch.py"
                
    
            
### Step 3 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 node list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/listener
/talker

```
                
    
            
### Step 4 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch demo_nodes_cpp talker_listener_launch.py
```
                
```bash
# StdIn - terminal 2
ros2 topic list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/chatter
/parameter_events
/rosout

```
                
    
            
### Step 5 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run demo_nodes_cpp add_two_ints_server
```
                
```bash
# StdIn - terminal 2
ros2 service list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically

```
                
    
            
### Step 6 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 run turtlesim turtlesim_node
```
                
```bash
# StdIn - terminal 2
ros2 param list
```
                
    

    
        
#### Expected Result
        
        
```bash
# StdOut - terminal 2
/turtlesim:
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time

```

knmcguire, 2024-05-03:
I will be testing this out on a raspberry pi 5 with Ubuntu Noble, and a ROS2 binary install with cyclonedds removed from the skip-tag.

knmcguire, 2024-05-03:
I'll be using this marklist to mark which ones work

1. help text

- [ ] ros2 -h
- [ ] ros2 --help
- [ ] ros2 topic -h
- [ ] ros2 topic --help

2.  tab completion

- [ ] ros2 tab tab
- [ ] ros2 lau tab 
- [ ] ros2 launch dem tab
- [ ] ros2 launch demo_nodes_ctab
- [ ] ros2 launch demo_nodes_cpp tab tab
- [ ] ros2 launch demo_nodes_cpp tatab
- [ ] ros2 launch demo_nodes_cpp talker_listener tab tab
- [ ] ros2 launch demo_nodes_cpp talker_listener_launch.p tab

3. node discovery

- [ ] ros2 launch demo_nodes_cpp talker_listener_launch.py & ros2 node list

4. Topic discovery

- [ ] ros2 launch demo_nodes_cpp talker_listener_launch.py & ros2 topic list

5.  service discovery

- [ ] ros2 run demo_nodes_cpp add_two_ints_server ros2 service list

6. step parameter discovery

- [ ] ros2 run turtlesim turtlesim_node & ros2 param list

knmcguire, 2024-05-03:
I started all terminals with
```
$ . ~/ros2_jazzy/ros2-linux/setup.bash
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

**All tests seem to pass**

clalancette, 2024-05-03:
Thanks for testing!

closed 2024-05-03 by clalancette